afollestad/material-camera #3: Add support for still image capture

tuhin10, 2015-11-24:


anvesh523, 2015-11-24:
Good one. 
 Suggestion for Image capture too ..... with 1:1 aspect ratio. It's very useful for all.

vanshg, 2016-01-19:
@afollestad Is there any intention of making this happen? If not, I will explore other options I have, but it would be great if everything was in one library

tomiu, 2016-02-29:
@afollestad
I can do that have had lots of experience of creating my own custom camera app already.

afollestad, 2016-02-29:
@tomiu that would be great, please test it as much as possible though.

This library was initially created for use in one of my employer's apps, so I don't want to break anything 😁

tomiu, 2016-03-03:
@afollestad Is there a reason that Camera2 is disabled by default?
Check CameraUtil, line 58

Sebas7ien, 2016-03-08:
+1 to take picture too

tomiu, 2016-03-22:
There's a fork here, with stillshot implemented:
https://github.com/tomiu/material-camera

afollestad, 2016-05-19:
I wouldn't recommend using that fork above since it is so outdated.

XiaolongQiao, 2016-05-27:
During the video recording, the picture is drawn. How to set

mowcixo, 2016-07-05:
@afollestad As I say in PR #70, I updated the @tomiu fork to the latest changes in master, added doc, and I think is a good idea to implement this PR, because is non intrusive and well made.

sugarmanz, 2016-08-11:
@afollestad: The still shot PR (#70) has since been tested by @tylerkrupicka and I. With our various bug fixes and @mowcixo s work updating the @tomiu fork, we have added still shot as a stable feature. More info #70].

afollestad, 2016-08-20:
In the latest release.

closed 2016-08-20 by afollestad

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.0

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.1