afollestad/material-camera #6: Not able to download library via Jitpack IO repository

jd-alexander, 2015-12-01:
Followed the instructions to add your library to my project but not able to download it. 
[image]

afollestad, 2015-12-01:
Well, the library is working fine for me, and JitPack reports that it built successfully: https://jitpack.io/#afollestad/material-camera

You might want to report an issue to JitPack here: https://github.com/jitpack/jitpack.io/issues

closed 2015-12-01 by afollestad

osflor, 2015-12-03:
I had the same problems at first until I realized there seems to be a tiny mistake in the instructions. They say: 
  compile('com.github.afollestad.material-camera:0.1.1@aar')
When it should probably be: 
  compile('com.github.afollestad:material-camera:0.1.1@aar') 
So just change the '**.**' to a '**:**' before the repo name and it works fine.

jd-alexander, 2015-12-03:
Oh thanks. I went through all the trouble of manually downloading the lib and adding it as a module.Glad I can go back to the normal way. :+1:

afollestad, 2015-12-04:
I also discovered the JitPack repository generally has to go in your root build.gradle file, rather than your module build.gradle file

jd-alexander, 2015-12-04:
Okay. The naming change of the compile statement you made to the readme
fixed it for me.