afollestad/material-camera #8: Camera preview is not good enough while rotating device....

anvesh523, 2015-12-03:
In my sony mobile While rotating portrait and landscape mode the preview is compressing and at 270* the preview going upside down ( reverse ) ..... | in sumsung mobile is the preview is compressing while switching portrait and landscape mode and any angle working normal....

i think you r using " android:layout_above="@+id/controlsFrame" " so the preview is compressing ..... check once

afollestad, 2015-12-03:
Hmm, the preview being upside down might be the result of the device reporting a wrong sensor orientation for the camera. There's not too much I can do since I don't have that device.

I'll look into the preview compression.

afollestad, 2016-01-13:
@anvesh523 can you check if this still happens with the latest version? A new method of playback is used, which is much more reliable (ExoMedia, which wraps around Google's ExoPlayer).

closed 2016-01-19 by afollestad

stephen-workpop, 2016-02-05:
This is happening on Galaxy S3. The preview is warped and the output is flipped. Try this code I wrote:

`/**
     * Rotate the camera's preview display orientation if necessary. This code is based off the
     * example in the documentation for {@link Camera#setDisplayOrientation(int)} as well as
     * https://github.com/commonsguy/cwac-camera/blob/master/camera/src/com/commonsware/cwac/camera/CameraView.java
     *
     * @param activity
     * @param cameraId
     * @param camera
     * @return  true if the aspect ratio should be flipped from w/h to h/w.
     */
    public static boolean setCameraDisplayOrientation(final Activity activity, final int cameraId, final Camera camera, final MediaRecorder mediaRecorder)
    {
        final int phoneRotation = activity.getWindowManager().getDefaultDisplay().getRotation();
        int degrees = 0;
        switch (phoneRotation)
        {
            case Surface.ROTATION_0:
                // A phone that has a naturally tall screen is held normally.
                degrees = 0;
                break;
            case Surface.ROTATION_90:
                // A phone that has a naturally tall screen is held in landscape.
                degrees = 90;
                break;
            case Surface.ROTATION_180:
                // A phone that has a naturally tall screen is held upside down.
                degrees = 180;
                break;
            case Surface.ROTATION_270:
                // A phone that has a naturally tall screen is held in landscape.
                degrees = 270;
                break;
        }

        int displayOrientation;
        final Camera.CameraInfo info = new Camera.CameraInfo();
        Camera.getCameraInfo(cameraId, info);
        if (info.facing == Camera.CameraInfo.CAMERA_FACING_FRONT)
        {
            displayOrientation = (info.orientation + degrees) % 360;
            // Compensate for the mirror.
            displayOrientation = (360 - displayOrientation) % 360;
        }
        else
        {
            displayOrientation = (info.orientation - degrees + 360) % 360;
        }

        camera.setDisplayOrientation(displayOrientation);

        int outputOrientation;
        if (activity.getRequestedOrientation() != ActivityInfo.SCREEN_ORIENTATION_UNSPECIFIED)
        {
            outputOrientation = getCameraPictureRotation(activity.getWindowManager().getDefaultDisplay().getOrientation(), cameraId);
        }
        else if (info.facing == Camera.CameraInfo.CAMERA_FACING_FRONT)
        {
            outputOrientation = (360 - displayOrientation) % 360;
        }
        else
        {
            outputOrientation = displayOrientation;
        }

        final Camera.Parameters params = camera.getParameters();
        params.setRotation(outputOrientation);
        camera.setParameters(params);

        mediaRecorder.setOrientationHint(outputOrientation);

        if ((displayOrientation == 90) || (displayOrientation == 270))
        {
            return true;
        }
        else
        {
            return false;
        }
    }

    /**
     * Based off of
     * https://github.com/commonsguy/cwac-camera/blob/master/camera/src/com/commonsware/cwac/camera/CameraView.java
     * which is based off of
     * http://developer.android.com/reference/android/hardware/Camera.Parameters.html#setRotation(int)
     *
     * @param orientation
     * @param cameraId
     * @return
     */
    private static int getCameraPictureRotation(int orientation, final int cameraId)
    {
        final Camera.CameraInfo info = new Camera.CameraInfo();
        Camera.getCameraInfo(cameraId, info);

        orientation = (orientation + 45) / 90 * 90;

        int rotation;

        if (info.facing == Camera.CameraInfo.CAMERA_FACING_FRONT)
        {
            rotation = (info.orientation - orientation + 360) % 360;
        }
        else
        {
            rotation = (info.orientation + orientation) % 360;
        }

        return rotation;
    }`

afollestad, 2016-02-06:
I use this library in multiple company apps and have had no issues during QA testing. 

But please submit a PR if you have fixes.