afollestad/material-camera #9: Setting time limit to videos

osflor, 2015-12-03:
Hi when I try to set a time limit for the video the countdown starts immediately, even before I press the record button. Is this the intended behavior? 

My code is:
new MaterialCamera(this)
                        .saveDir(saveFolder)
                        .primaryColorAttr(R.attr.colorPrimary)
                        .showPortraitWarning(true)
                        .autoSubmit(false)
                        .defaultToFrontFacing(false)
                        .lengthLimitSeconds(10)
                        .start(REQUEST_VIDEO);
Device: Nexus 5
Android Version: 6.0

afollestad, 2015-12-03:
Currently, it is, because of how the library is used in one of my employer's apps. I can add a setting to change the behavior.

osflor, 2015-12-03:
If you ever do that that would be great. Btw thanks for an awesome library.

afollestad, 2015-12-03:
When the next update is available, it will work like this:

```java
new MaterialCamera(this)
    .countdownMinutes(2.5f)
    .countdownImmediately(true)
    .start(CAMERA_RQ);
```

If you don't use `countdownImmediately(true)`, it will wait until the user presses 'Record' to start the countdown.

closed 2015-12-03 by afollestad