afollestad/material-camera #15: The front camera in lolipop 5.1 is flip backwards and mirrored

destrike, 2016-01-14:
[image]

I don't test it on other version yet because I only got 5.1 lolipop android in my hands. Hope this help.

afollestad, 2016-01-14:
Hmm, that doesn't happen to me on a Nexus 6, Nexus 5X, Nexus 6P, or Nexus 9 with 5.1 or 6.0.1.

destrike, 2016-01-18:
I'm tried on nexus 10 and it's still flipped, Thanks for your reply, gonna test it on my mobile device

afollestad, 2016-02-09:
Sorry for the late response.

Was this with the current live version? Or with the latest code? There have been a lot of changes in the code base that haven't been released for quite a while.

destrike, 2016-02-10:
It's on the current version.

tomiu, 2016-03-01:
@afollestad  Might have an idea. Will try it during the weekend

joomi, 2016-03-04:
I have tested the code on both Galaxy S6 (Android 5.1.1) and LG g3 (Android 5.0)
and on both the front camera video is mirrored.
The bigger issue is with the LG g3, the video quality is very bad (somehow chooses a very low resolution) and the video is scaled and changing proportions (distorted)

afollestad, 2016-03-04:
Make sure you are using a newer version.

---

```Gradle
dependencies {
    ...
    compile('com.github.afollestad:material-camera:bda3fd2ce1@aar') {
        transitive = true
    }
}
```

joomi, 2016-03-04:
I have used this version
com.github.afollestad:material-camera:d2229233c4@aar
But I still get the same issues with the newest version (bda3fd2ce1)
Using my LG G3

patelkarn, 2016-05-03:
I can confirm that Nexus 10 does have that problem. After doing some searching and debugging I found out that when device is in landscape and we call the function isPortrati(Context activity) to check if its in portrait or not it return true. I found out that some devices are meant to be used in landscape like Nexus 10 therefore camera returns DEGREE_0 while in landscape (I could be wrong but that's my guess). This also causes problem when we actually try to record the video the orientation does not get set properly because of the getActivityOrientation(@NonNull Activity context) function
Temporarily I fixed this in Degree.java class as follows (maybe you guys can come up with a better solution):
```java
public static boolean isPortrait(Context activity){ 
 return activity.getResources().getConfiguration().orientation == Configuration.ORIENTATION_PORTRAIT;
//	    isPortrait(getDisplayRotation(activity));
    }
```
```java
    @ActivityOrientation
    public static int getActivityOrientation(@NonNull Activity context) {
	    WindowManager windowManager = (WindowManager) context.getSystemService(Context.WINDOW_SERVICE);
	    Configuration configuration = context.getResources().getConfiguration();
        @DegreeUnits
        final int rotation = windowManager.getDefaultDisplay().getRotation();
//        final int rotation = getDisplayRotation(context);
        switch (rotation) {
            case Surface.ROTATION_0:
	            if(configuration.orientation == Configuration.ORIENTATION_PORTRAIT)
                    return ActivityInfo.SCREEN_ORIENTATION_PORTRAIT;
	            else
		            return ActivityInfo.SCREEN_ORIENTATION_UNSPECIFIED;
            case Surface.ROTATION_90:
	            if(configuration.orientation == Configuration.ORIENTATION_PORTRAIT)
                    return ActivityInfo.SCREEN_ORIENTATION_LANDSCAPE;
	            else
		            return ActivityInfo.SCREEN_ORIENTATION_UNSPECIFIED;
            case Surface.ROTATION_180:
	            if(configuration.orientation == Configuration.ORIENTATION_PORTRAIT)
                    return ActivityInfo.SCREEN_ORIENTATION_REVERSE_PORTRAIT;
	            else
		            return ActivityInfo.SCREEN_ORIENTATION_UNSPECIFIED;
            case Surface.ROTATION_270:
	            if(configuration.orientation == Configuration.ORIENTATION_LANDSCAPE)
                    return ActivityInfo.SCREEN_ORIENTATION_REVERSE_LANDSCAPE;
	            else
	                return ActivityInfo.SCREEN_ORIENTATION_UNSPECIFIED;
            default:
                Log.e("Degrees", "Unknown screen orientation. Defaulting to portrait.");
	            if(configuration.orientation == Configuration.ORIENTATION_PORTRAIT)
                    return ActivityInfo.SCREEN_ORIENTATION_PORTRAIT;
	            else
		            return ActivityInfo.SCREEN_ORIENTATION_UNSPECIFIED;
        }
    }
```
in the above function when it return ActivityInfo.SCREEN_ORIENTATION_UNSPECIFIED then i dont change the orientation of the device.
```java
    public boolean startRecordingVideo() {
        if (mInterface != null && mInterface.hasLengthLimit() && !mInterface.countdownImmediately()) {
            // Countdown wasn't started in onResume, start it now
            if (mInterface.getRecordingStart() == -1)
                mInterface.setRecordingStart(System.currentTimeMillis());
            startCounter();
        }

	    final int orientation = Degrees.getActivityOrientation(getActivity());
	    if (orientation != -1)
		    getActivity().setRequestedOrientation(orientation);

        mInterface.setDidRecord(true);
        return true;
    }
```
Hopefully the information helps

afollestad, 2016-05-09:
Unfortunately a lot of those workaround solutions will cause issues on potentially hundreds of other devices. Any changes like that would have to be tested on other devices before we put it into the code base.

patelkarn, 2016-06-02:
Completely agree on that I tested this on Nexus 6P, Nexus 5X, Nexus 10 works fine on those devices.

closed 2016-06-02 by afollestad