afollestad/material-camera #18: Quality: resolution low, bitrate high

ManicMapple, 2016-01-26:
Hi,
This is the codec info from a file produced by Material Camera 
* Device: Nexus 5

```bash
  Duration: 00:00:10.92, start: 0.000000, bitrate: 2261 kb/s
    Stream #0:0(eng): Video: h264 (Baseline) (avc1 / 0x31637661), yuv420p, 640x480, 2235 kb/s, 21.72 fps, 49.92 tbr, 90k tbn, 180k tbc (default)
    Metadata:
      creation_time   : 2016-01-26 13:35:36
      handler_name    : VideoHandle
    Stream #0:1(eng): Audio: aac (mp4a / 0x6134706D), 48000 Hz, mono, fltp, 118kb/s (default)
```

Wow 2235 kb/s Bitrate at a 640x480@22.1 Fps

So questions: How do i set the resolution, bitrate and fps? I cold find this in Camera2Fragment https://github.com/afollestad/material-camera/blob/244ff8763365e7bf9985ba9e0c429b8b92797a3e/library/src/main/java/com/afollestad/materialcamera/internal/Camera2Fragment.java#L443 but why is this hardcoded?

afollestad, 2016-01-26:
This is on purpose, this library is used in one of my company's apps which requires those settings.

But I will make it configurable.

closed 2016-01-26 by afollestad

ManicMapple, 2016-01-27:
Hi afollestad,
thanks for the fast update. I am still having Problems with your fallback solution in  https://github.com/afollestad/material-camera/blob/c7fab31af5d5f77ef93ae25153bcd867db0a1a47/library/src/main/java/com/afollestad/materialcamera/internal/CameraFragment.java#L55

When there is no match, then in my case the backup resolution is the worst possible option 177x144 because it is at the end of the list of sizes. Is this normal behaviour? And why do you use previwe sizes here? Wouldn't  getSupportedVideoSizes be the better choice here? Problem is that some devices can record e.g. 1080p but do not have a >= 1080p display.

Here I added an aditional check to fallback from 852x480 to 720x480 and not to 177x144

```java
    private static Camera.Size chooseVideoSize(BaseCaptureInterface ci, List<Camera.Size> choices) {
        Camera.Size backupSize = null;
        for (Camera.Size size : choices) {
            if (size.height <= ci.videoPreferredHeight()) {
                if (size.width == size.height * ci.videoPreferredAspect())
                    return size;
                
                if(ci.videoPreferredHeight() == size.height) {
                    backupSize = size;
                }
            }
        }
    }
```

Anyway, thanks for your effort and sharing of code.

afollestad, 2016-01-27:
@ManicMapple thanks for the feedback!

I've added that additional check to both `CameraFragment` and `Camera2Fragment`. `CameraFragment` falls back to the preview sizes *if* `getSupportedVideoSizes()` returns a null or empty list.

afollestad, 2016-04-11:
https://github.com/afollestad/material-camera/releases/tag/0.2.2

commit 4fc3bdd9 (2016-01-26): Added settings for video bit rate, frame rate, preferred height, and preferred aspect ratio. Upgraded Gradle plugin and ExoMedia. Resolves #18.