afollestad/material-camera #19: video is distorted

joomi, 2016-02-12:
Hi Aidan 
First I must say you did a great job, it looks really good, thanks.
Trying it out, I find that the videos are distorted and lost their original scale.
I wonder if you encounter this issue.
Im using LG G3
android version 5

EzequielAdrianM, 2016-03-10:
I've experienced distorted video while recording ONLY with Front camera. But It is only on the recording preview, since the saved video is fine. Also this only happens in Portrait mode.

joomi, 2016-03-10:
In my case it's also in the saved video, not just preview

lezardos, 2016-03-15:
The preview is distorted on Nexus 4 as well.
Regards

afollestad, 2016-04-18:
0.2.5 should help.

https://github.com/afollestad/material-camera/releases/tag/0.2.5

closed 2016-04-18 by afollestad