afollestad/material-camera #24: Build finished with Errors.

EzequielAdrianM, 2016-03-08:
When I add all your dependencies to my Project causes the following error:

    ...\app\build\intermediates\exploded-aar\com.android.support\appcompat-v7\23.2.0\res\values-v23\values-v23.xml
    Error:(2) Error retrieving parent for item: No resource found that matches the given name 'android:TextAppearance.Material.Widget.Button.Inverse'.
    Error:(2) Error retrieving parent for item: No resource found that matches the given name 'android:Widget.Material.Button.Colored'.

My project compileSdkVersion is 22 and buildtoolsversion is 22.0.1 and targetSdkVersion is 22.
MinSdkversion is 15.

afollestad, 2016-03-08:
You need to use AppCompat 23.2.0 in your app.

closed 2016-03-08 by afollestad

afollestad, 2016-04-11:
https://github.com/afollestad/material-camera/releases/tag/0.2.2