afollestad/material-camera #26: throw ResouecesNotFoundException

ShinJoonSeob, 2016-03-09:
Hi!

i will your library using on my app.

but i shown the below messages.

FATAL EXCEPTION: main
                                                             Process: com.ktc.seals, PID: 6859
                                                             java.lang.RuntimeException: Unable to start activity ComponentInfo{com.ktc.seals/com.afollestad.materialcamera.CaptureActivity}: android.content.res.Resources$NotFoundException: File res/drawable-xxhdpi-v4/mcam_camera_rear.png from xml type xml resource ID #0x7f0200af

do you have any solutions?

thanks

afollestad, 2016-03-09:
That file doesn't exist since newer versions of the library are using Vector drawables. You need AppCompat 23.2.0.

closed 2016-03-09 by afollestad

afollestad, 2016-04-11:
https://github.com/afollestad/material-camera/releases/tag/0.2.2

afollestad, 2016-04-17:
https://github.com/afollestad/material-camera/releases/tag/0.2.4