afollestad/material-camera #28: Samsung Galaxy S3 issue

ruha9005, 2016-03-10:
FATAL EXCEPTION: main
java.lang.RuntimeException: Unable to start activity ComponentInfo{ru.ntc_vulkan.rberozov.transcrypto/com.afollestad.materialcamera.CaptureActivity}: android.content.res.Resources$NotFoundException: File res/drawable-xhdpi-v4/mcam_camera_rear.png from xml type xml resource ID #0x7f02006f
                                                                                        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2100)
....
App crashes when start MaterialCamera

aiis, 2016-03-10:
I ran into this problem just a few minutes ago. For me, it was an issue with version 23.2.0 of the support library (which I updated to last night).

http://stackoverflow.com/questions/35622438/update-android-support-library-to-23-2-0-cause-error-xmlpullparserexception-bin

I was able to fix my problem by using the accepted answer at the link above.

ruha9005, 2016-03-16:
Thanks, it fixed my problem. Please, add this information to Readme.md for other developers.

afollestad, 2016-04-11:
The library is being updated for 23.3.0 of the support library which solves that issue.

closed 2016-04-11 by afollestad

afollestad, 2016-04-11:
https://github.com/afollestad/material-camera/releases/tag/0.2.2

uzman, 2016-04-12:
hi, the latest version has still same problem for Galaxy S3.
i tried all here: http://stackoverflow.com/questions/35622438/update-android-support-library-to-23-2-0-cause-error-xmlpullparserexception-bin

bun no luck.
i updated studio the latest building. i made gradle 2.0.
my settings:
        classpath 'com.android.tools.build:gradle:2.0.0'

    compileSdkVersion 23
    buildToolsVersion "23.0.3"

    defaultConfig {
        applicationId "com.qqqqqq"
        minSdkVersion 15
        targetSdkVersion 23
        versionCode 1
        versionName "1.0"
        multiDexEnabled true
        vectorDrawables.useSupportLibrary = true
    }
    compile 'com.android.support:design:23.3.0'

it still gives NotFoundException for mcam_camera_rear.png 
can anyone help?

Corical, 2016-09-28:
Same issue here on a Samsung Galaxy GT-I9190 (S4 mini) running 4.4.2

ruha9005, 2016-10-27:
@Corical @uzman 
I use this build that fix this error and waiting for 0.4.2 release.
`compile 'com.github.afollestad:material-camera:8cd8c477b7'`