afollestad/material-camera #30: Video is not saving correctly

ryanzor, 2016-03-13:
Using a Samsung S6 when I take a video with material camera on the preview instead of being able to save the video I can only see a still image.  When I do save the video I get the same effect (I see there is a video there, but when I play the video the image doesn't change).

I got this both in my own app and the sample app using

    compile('com.github.afollestad:material-camera:82aea91013@aar') {
        transitive = true
    }

EzequielAdrianM, 2016-03-15:
I have the same problem. Samsung S4 API 22. Video preview is still image and saved video is also still image. Audio is recorded correctly.

ryanzor, 2016-03-19:
I'm thinking this problem is only the result of the documentation needing updating.  I used:

    compile 'com.github.afollestad:material-camera:0.2.0'

and that fixed this issue.  Hoping for a documentation update soon :)

closed 2016-04-11 by afollestad

afollestad, 2016-04-11:
https://github.com/afollestad/material-camera/releases/tag/0.2.2

canbakis, 2016-06-02:
Hi, we are having the same issue on com.afollestad:material-camera:0.2.13
The device is Samsung S6 Edge

juliobcosta, 2016-12-12:
Hi... I have the same problem with the last version 0.4.4
My device also Samsung S6 Edge. How can I fix this.

Thank you very much

Dhrupal, 2017-06-09:
Issues in samsung galxy s7
1) video length is not proper
2) video playing at the end of the length
3) sometimes video recording fail and open camera again automatically.