afollestad/material-camera #39: Pressing Back button while recording

uzman, 2016-04-11:
if the user press back button while recording, screen goes black and suspends but it still records background. i checked the file on sd card. (it recorded image+sound until pressing back, and just sound was recorded after pressing back.) 
i tried with Galaxy S3 + 4.4.2.
i didn't check with other device and android versions.

is it possible to stop recording if user presses the back button or disable back event?

here is my settings;

        new MaterialCamera(this)
                .saveDir(saveDir)
                .defaultToFrontFacing(true)
                .showPortraitWarning(false)
                .autoSubmit(true)
                .allowRetry(false)
                .countdownSeconds(30f)
                .countdownImmediately(false)
                .primaryColorAttr(R.attr.colorPrimary)
                .start(CAMERA_RQ);

afollestad, 2016-04-11:
https://github.com/afollestad/material-camera/releases/tag/0.2.2

closed 2016-04-11 by afollestad

afollestad, 2016-04-17:
https://github.com/afollestad/material-camera/releases/tag/0.2.4

commit 5fd60739 (2016-04-11): Resolved #39