afollestad/material-camera #41: NotFoundException res/drawable-xhdpi-v4/mcam_camera_rear.png

uzman, 2016-04-12:
Hi,
im testing it with Galaxy S3 + 4.4.2
it was working before but the latest build gives error when i try to open camera.
**NotFoundException for res/drawable-xhdpi-v4/mcam_camera_rear.png **
i read the solution but it doesn't work.

here is the gradle settings:

classpath 'com.android.tools.build:gradle:2.0.0'

compileSdkVersion 23
buildToolsVersion "23.0.3"

defaultConfig {
    applicationId "com.qqqqqq"
    minSdkVersion 15
    targetSdkVersion 23
    versionCode 1
    versionName "1.0"
    multiDexEnabled true
    vectorDrawables.useSupportLibrary = true
}
compile 'com.android.support:design:23.3.0'

    new MaterialCamera(this)
            .saveDir(saveDir)
            .defaultToFrontFacing(true)
            .showPortraitWarning(false)
            .autoSubmit(true)
            .allowRetry(false)
            .countdownSeconds(30f)
            .countdownImmediately(false)
            .primaryColorAttr(R.attr.colorPrimary)
            .start(CAMERA_RQ);

afollestad, 2016-04-12:
As long as you're using version 23.3.0 of AppCompat and the Support Library too, there should be no problems.

You should also remove `vectorDrawables.useSupportLibrary = true`, that has no use with 23.3.0.

uzman, 2016-04-12:
No,
i used 23.0.0 appcompat and support library but still same error.
i deleted vectorDrawables.useSupportLibrary = true.

dependencies {
    compile fileTree(dir: 'libs', include: ['*.jar'])
    compile 'com.android.support:appcompat-v7:23.3.0'
    compile 'com.android.support:support-v13:23.3.0' // i tried with it and without it too.
    compile 'com.android.support:design:23.3.0'
    compile 'com.google.android.gms:play-services-gcm:8.3.0'
    compile 'com.google.android.gms:play-services-analytics:8.3.0'
    compile 'com.google.android.gms:play-services-ads:8.3.0'
    compile 'com.google.android.gms:play-services-appindexing:8.3.0'
    compile('com.github.afollestad:material-camera:0.2.2@aar') {
        transitive = true
    }
}

afollestad, 2016-04-13:
Are you using a newer version of Build Tools, like 23.0.3?

uzman, 2016-04-13:
yes,
    compileSdkVersion 23
    buildToolsVersion "23.0.3"

i couldn't solve it.

closed 2016-04-17 by afollestad

afollestad, 2016-04-17:
https://github.com/afollestad/material-camera/releases/tag/0.2.4

commit 1645d30b (2016-04-17): Resolved #41