afollestad/material-camera #42: Aspect Ratio 1:1

JorgeBazarte, 2016-04-15:
First, Great Job!!
How can I set an aspect ratio 1:1? I've been trying with different configuration but always fails and give me a file with the next qualities, 176x144.
I'm developing an app that includes small videos (15 seconds)

ruha9005, 2016-04-15:
try it:
`.videoPreferredAspect(1f / 1f)`

gouravd, 2016-04-15:
I do not think this will work, without significant changes to the source. According to the current source, only supported Aspect ratios by the android device (like 3:4 and 9:16) will work.

afollestad, 2016-04-15:
There is no way to force a specific aspect ratio, thus why you can only set a "preferred" aspect ratio. 

If you use an aspect ratio or output size not reported as "supported" by the device, your app will crash.

closed 2016-04-15 by afollestad