afollestad/material-camera #44: CameraUtil.hasCamera2 API judge issue

puyanlin, 2016-04-18:
```
final CameraCharacteristics characteristics = manager.getCameraCharacteristics(str);
final int supportLevel = characteristics.get(CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL);
if (supportLevel == CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_LEGACY) {
       notNull = false;
       break;
}

```
there is a problem in the 'if' statement, some old but updatable devices just support Camera2 API, but its system parameter just match 'supportLevel == CameraCharacteristics.INFO_SUPPORTED_HARDWARE_LEVEL_LEGACY' condition, so they would use incorrect ratio old camera API... ex: Nexus 4/Nexus 7(2rd-gen)...

closed 2016-04-18 by afollestad

puyanlin, 2016-05-10:
why add these code back in v 0.2.6?

afollestad, 2016-05-10:
The change that removed it broke a lot for other devices.

commit 75d5ccbe (2016-04-18): Resolved #44