afollestad/material-camera #46: Preview is stretched on some devices

tuandt-gem, 2016-04-25:
I run on Sony Xperia M4 Aqua Dual E2312 and preivew screen looks like stretched.

Could you fix it?
[image]

ManicMapple, 2016-04-25:
Unfortunately this also seems to effect the resulting video. I have set
.videoPreferredHeight(720)
.videoPreferredAspect(16f / 9f)

and it results in a stretched video on at least
motorolla g2
samsung galaxy s4
samsung galaxy s7

Until recently I used the commit https://github.com/afollestad/material-camera/commit/8f51691d2e5fdba1efb54befd9ed12e3649926e4 where this problem did not occur and checked every commit after that for changes that might affect this -  but the only thing I can think of is an update in one of the libs (material dialogs maybe?)

afollestad, 2016-04-26:
@ManicMapple my Material Dialogs library couldn't have any affect on video previews.

ManicMapple, 2016-04-28:
@afollestad I guess the problem is the surface. http://stackoverflow.com/questions/22105251/android-mediarecorder-aspect-ratio-incorrect

The cases where I had Problems were when preview size (chooseOptimalSize()) and videosize (chooseVideoSize()) had different aspects. (1280x720 vs 960x720)

I solved teh problem (for me, badly) by using the video size as preview size...  This way the videos had the correct aspect.

Sadly, on samsung galaxy s7 thsi dows not help... the video is just a still image.

afaucogney, 2016-04-29:
Same issue on my Galaxy S4 Advanced

closed 2016-05-09 by afollestad

afollestad, 2016-05-09:
https://github.com/afollestad/material-camera/releases/tag/0.2.6

commit 4b004b79 (2016-05-09): Resolved #46