afollestad/material-camera #48: Emtpy video files when switch camera.

vxhviet, 2016-04-26:
Hi @afollestad, thank you for your awesome libs.

Currently, when I flip the camera back and forth there's a lot of empty video files get produced. For example, for this sequence:

Open up camera > Back to Front > Front to Back > Back to Front > Front to Back > Record

I get these files:
VID_20160426_091310.mp4 (0B)
VID_20160426_091314.mp4 (0B)  
VID_20160426_091320.mp4 (0B)
VID_20160426_091325.mp4 (0B)
VID_20160426_091330.mp4 (371.53KB)

My code snippet using version 0.2.5 is as below:

```java
new ModifiedMaterialCamera(context)                      
                .allowRetry(true)                         
                .autoSubmit(true)                         // Whether or not user is allowed to playback videos after recording. This can affect other things, discussed in the next section.
                .saveDir(saveFolder)                       // The folder recorded videos are saved to
                .primaryColorRes(R.color.myucrop_color_widget_background)     // The theme color used for the camera, defaults to colorPrimary of Activity in the constructor
                .showPortraitWarning(false)                 // Whether or not a warning is displayed if the user presses record in portrait orientation
                .defaultToFrontFacing(false)               // Whether or not the camera will initially show the front facing camera
                .retryExits(false)                         // If true, the 'Retry' button in the playback screen will exit the camera instead of going back to the recorder
                .restartTimerOnRetry(false)                // If true, the countdown timer is reset to 0 when the user taps 'Retry' in playback
                .continueTimerInPlayback(false)            // If true, the countdown timer will continue to go down during playback, rather than pausing.
                .videoBitRate(1024000)                     // Sets a custom bit rate for video recording.
                .videoFrameRate(30)                        // Sets a custom frame rate (FPS) for video recording.
                .videoPreferredHeight(720)                 // Sets a preferred height for the recorded video output.
                .videoPreferredAspect(16f / 9f)             // Sets a preferred aspect ratio for the recorded video output.
                .maxAllowedFileSize(1024 * 1024 * 5)       // Sets a max file size of 5MB, recording will stop if file reaches this limit. Keep in mind, the FAT file system has a file size limit of 4GB.
                .start(CAMERA_RQ);
```

I've modified your library a bit, but only to customized the look (changing icon, color, etc...). I haven't touched any of the logic.

So what do you think would possibly causing this bug?

afaucogney, 2016-04-29:
Did you use the retry feature ?

vxhviet, 2016-04-29:
Well, according to my snippet `allowRetry(true)` then I guess yes.
Is it the main culprit, if so, I can happily leave it out.

vxhviet, 2016-04-29:
Also, I would like to add, open the camera then go back without recording anything also produce empty video file.

afaucogney, 2016-04-29:
On my side, I have the impression this happen when I first get a video, then do not save it and retry a new one. But that's just an impression.

vxhviet, 2016-04-29:
Could you check the output folder? In my output folder, there is literally 0 byte video file which can not be opened by any player.

afaucogney, 2016-04-29:
the same for me. I have a couple of void videos.

closed 2016-05-09 by afollestad

afollestad, 2016-05-09:
https://github.com/afollestad/material-camera/releases/tag/0.2.6

commit 6241560a (2016-05-09): Parameters are set in a different way, resolving a lot of errors and making default video quality a lot better. Resolved #50, resolved #48, resolved #47, resolved #45.