afollestad/material-camera #50: It closes immediately when i press the record button.

uzman, 2016-05-06:
hi,
i tried with latest (0.2.5) and 0.2.4 but there is weird bug. it closes the camera immediately if i press the record button. but there is no crash.
it can be related with output format.

Device: Samsung S3 + v 4.4.2

settings:
        new MaterialCamera(this)
                .saveDir(saveDir)
                .defaultToFrontFacing(true)
                .showPortraitWarning(false)
                .autoSubmit(true)
                .allowRetry(false)
                .countdownSeconds(30f)
                .countdownImmediately(false)
                .primaryColorAttr(R.attr.colorPrimary)
                .start(CAMERA_RQ);

logcat:

E/MediaRecorder: start failed: -19
java.lang.RuntimeException: start failed.
at android.media.MediaRecorder.start(Native Method)
at com.afollestad.materialcamera.internal.CameraFragment.startRecordingVideo(CameraFragment.java:360)
at com.afollestad.materialcamera.internal.CameraFragment.onClick(CameraFragment.java:133)
at android.view.View.performClick(View.java:4640)
at android.view.View$PerformClick.run(View.java:19421)
at android.os.Handler.handleCallback(Handler.java:733)
at android.os.Handler.dispatchMessage(Handler.java:95)
at android.os.Looper.loop(Looper.java:136)
at android.app.ActivityThread.main(ActivityThread.java:5476)
at java.lang.reflect.Method.invokeNative(Native Method)
at java.lang.reflect.Method.invoke(Method.java:515)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1283)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1099)
at dalvik.system.NativeStart.main(Native Method)

uzman, 2016-05-06:
i found it.
frame rate must be set.
                .videoFrameRate(30)

it works now. but there is still same bug. pressing back button freezes the app while recording. 
(https://github.com/afollestad/material-camera/issues/39)

closed 2016-05-09 by afollestad

afollestad, 2016-05-09:
https://github.com/afollestad/material-camera/releases/tag/0.2.6

commit b3d7d7fc (2016-05-09): Parameters are set in a different way, resolving a lot of errors and making default video quality a lot better. Resolved #50, resolved #47, resolved #45.
commit 6241560a (2016-05-09): Parameters are set in a different way, resolving a lot of errors and making default video quality a lot better. Resolved #50, resolved #48, resolved #47, resolved #45.