afollestad/material-camera #52: defaultToFrontFacing method doesn't seem to do anything

upta, 2016-05-11:
It seems that using the defaultToFrontFacing method with either true or false results in the same thing:  the camera starts up looking me straight in the face ;)

I've tested this on 3 devices:

- Nexus 5X (6.0.1)
- Nexus 5 (6.0.0)
- Galaxy S4 (4.4.2)

Behavior is exactly the same on all three and switches between front and back cameras just fine by the button press, but I can't it to start using the back camera.

Thanks!

afollestad, 2016-05-11:
@upta I'm unable to reproduce any issues on my Nexus 6P with Android N.

Make sure you're using the latest version of the library. It's on jCenter not by the way, rather than JitPack.io.

closed 2016-05-11 by afollestad

upta, 2016-05-11:
Hmm, I'd had 0.2.5, switched to 0.2.6 (which I had to leave the jitpack.io reference in there for material-dialogs, just fyi) and I still get the same issue with my devices :(

Not the end of the world, just kinda odd that it's doing it with all three on different android versions