afollestad/material-camera #55: 401 error

IFDeveloper, 2016-05-21:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

compile 'com.afollestad:material-camera:0.2.12' 
results in :
Error:Could not GET 'https://jitpack.io/com/afollestad/easyvideoplayer/0.2.6/easyvideoplayer-0.2.6.pom'. Received status code 401 from server: Repo not found or no access token provided

tbarsbay, 2016-05-21:
I'm having this issue as well.

afollestad, 2016-05-22:
The library isn't on JitPack, it's on jCenter and Bintray.

afollestad, 2016-05-22:
See the README.

closed 2016-05-22 by afollestad

tbarsbay, 2016-05-22:
I'm using jcenter in my build file but still receiving the error described.

afollestad, 2016-05-22:
See https://github.com/afollestad/material-camera/issues/56

IFDeveloper, 2016-05-22:
issue solved
pls add 
maven { url "https://dl.bintray.com/drummer-aidan/maven" } to README file 
since dependency is not so obvious. thx

afollestad, 2016-05-22:
I won't add it to the README, that's only a work around. Not a solution to why it's not resolving from jCenter correctly.

IFDeveloper, 2016-05-22:
ok, w8ing for solution, thx anyway

amouly, 2016-05-24:
Same issue for me, with latest build tools and Studio versions. Fixed with @IFDeveloper comment.

Sateesh2, 2016-06-21:
still getting 401 error