afollestad/material-camera #58: Crash on video record button click not able to record video

chetanurmaliya, 2016-05-28:
It opens the camera preview but crashed on video record button click.

 - Material Camera version: `0.2.13`
 - Device OS version: `5.1.1`
 - Device Manufacturer: `Samsung`
 - Device Name: `Galaxy E7`

DenisBronx, 2016-06-15:
Hi i had the same issue with a Samsung Galaxy s4 and i solved it by setting the preferred aspect to 16f / 9f (videoPreferredAspect(16f / 9f))

ruudschouten, 2016-06-17:
I get the same error

    Material Camera version: 0.2.13
    Device OS version: 6.0.1 (CyanogenMod)
    Device Manufacturer: Motorola
    Device Name: Moto G 2014 LTE

Log:
> 
06-17 10:50:01.316 17594-17594/com.spinnov.remotecamera E/MediaRecorder: start failed: -19
06-17 10:50:01.319 17594-17594/com.spinnov.remotecamera W/System.err: java.lang.RuntimeException: start failed.
06-17 10:50:01.323 17594-17594/com.spinnov.remotecamera W/System.err:     at android.media.MediaRecorder.start(Native Method)
06-17 10:50:01.323 17594-17594/com.spinnov.remotecamera W/System.err:     at com.afollestad.materialcamera.internal.CameraFragment.startRecordingVideo(CameraFragment.java:366)
06-17 10:50:01.323 17594-17594/com.spinnov.remotecamera W/System.err:     at com.afollestad.materialcamera.internal.BaseCameraFragment.onClick(BaseCameraFragment.java:272)
06-17 10:50:01.323 17594-17594/com.spinnov.remotecamera W/System.err:     at com.afollestad.materialcamera.internal.CameraFragment.onClick(CameraFragment.java:138)
06-17 10:50:01.323 17594-17594/com.spinnov.remotecamera W/System.err:     at android.view.View.performClick(View.java:5204)
06-17 10:50:01.323 17594-17594/com.spinnov.remotecamera W/System.err:     at android.view.View$PerformClick.run(View.java:21158)
06-17 10:50:01.324 17594-17594/com.spinnov.remotecamera W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
06-17 10:50:01.324 17594-17594/com.spinnov.remotecamera W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
06-17 10:50:01.324 17594-17594/com.spinnov.remotecamera W/System.err:     at android.os.Looper.loop(Looper.java:148)
06-17 10:50:01.324 17594-17594/com.spinnov.remotecamera W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5461)
06-17 10:50:01.324 17594-17594/com.spinnov.remotecamera W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
06-17 10:50:01.324 17594-17594/com.spinnov.remotecamera W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
06-17 10:50:01.324 17594-17594/com.spinnov.remotecamera W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
06-17 10:50:01.488 17594-17864/com.spinnov.remotecamera D/OpenGLRenderer: endAllStagingAnimators on 0xb90a6188 (RippleDrawable) with handle 0xb8ed0aa8

Still happens when setting preferred aspect to 16f/9f as suggested by DenisBronx.

ruudschouten, 2016-06-17:
I was able to resolve this issue by setting FrameRate to 30.

afollestad, 2016-06-26:
Unfortunately different devices, especially Samsung devices, like to behave differently with default settings.

This crash only happens on Samsung devices, but it could break the library for other devices.

Dhrupal, 2017-08-22:
I am facing the same issue in One plus 4, os 7. Anyone have got solution?