afollestad/material-camera #61: Samsung Galaxy Note 5 does not record video properly.

patelkarn, 2016-06-02:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.2.9`
 - Device OS version: `5.1.1`
 - Device Manufacturer: `Samsung`
 - Device Name: `Samsung Galaxy Note 5`

So let me describe the issue first: What happens is the video gets recorded fine however when playing it back it would play the audio but no video. And once the seek bar reaches the end it would start playing the video. This happen quite often on Samsung Galaxy Note 5. The only way to fix this issue is to restart the device and it would start recording fine. However, eventually it would go bad.

Any suggestion or has anyone experienced this issue?

patelkarn, 2016-06-03:
Also happens on Samsung Galaxy Note Android 6.0

DenisBronx, 2016-06-15:
Same issue here with Samsung Galaxy s6

afollestad, 2016-06-26:
Not much I can do on my end to fix this, it only happens on Samsung devices.

Someone else will probably have to find a fix that doesn't break it for other devices.

patelkarn, 2016-07-12:
Forcing to legacy camera fixes the issue.

UglyTroLL, 2016-09-03:
@patelkarn How can you do it from the code?

patelkarn, 2016-09-04:
@UglyTroLL 
If you look at the MaterialCamera class there is forceCamera1() function which will use the legacy camera api.
```
    public MaterialCamera forceCamera1() {
        mForceCamera1 = true;
        return this;
    }
```
Just call
```
new MaterialCamera(this)
    ...... 
    .forceCamera1()        // This will force to use legacy camera api
    .start(CAMERA_RQ); 
```

sealedtx, 2016-12-01:
I had simillar problem, but i found this SO answer - http://stackoverflow.com/a/38268627/6046240, in my case it was on Samsung Galaxy 6 and Galaxy 7, the problem was that first frame took place too much time. I used mp4parser for java to fix it's length manually - https://github.com/sannies/mp4parser

tylerkrupicka, 2016-12-01:
That is quite a weird bug. the Camera 1 API is forced in stillshot for Samsung devices due to Camera 2 inconsistencies. It might be worth testing this for video as well. @sugarmanz has had this problem on his S6 so maybe he could try testing workarounds now that we have identified the issue.

Dhrupal, 2017-06-09:
@patelkarn Your legacy camera issue solved this issue.