afollestad/material-camera #62: Low video quality

Max0n, 2016-06-05:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.2.13`
 - Device OS version: `5.1`
 - Device Manufacturer: `Meizu`
 - Device Name: `m2 note`

I just redacted a part of the MainActivity.java.

sfilmak, 2016-06-06:
You need to open BaseCaptureActivity.java in Material Camera Library and change video preferred aspect from 4f / 3f to 16f / 9f.

Max0n, 2016-06-07:
Maybe in MainActivity.java not in BaseCaptureActivity.java? I can't find video preferred activity in this but it's in MainActivity.java.

Max0n, 2016-06-07:
... And it has not given result too. Video becomes blinking and quality becomes grainy. I hope you understand me because I'm Russian and bad speak English.

sfilmak, 2016-06-07:
Тогда, пожалуй, нам лучше перейти на русский

sfilmak, 2016-06-07:
Нет, все верно, эту строчку можно найти в файле BaseCaptureActivity:

`  @Override
    public float videoPreferredAspect() {
        return getIntent().getFloatExtra(CameraIntentKey.VIDEO_PREFERRED_ASPECT, 4f / 3f);
    }

    @Override
    public int videoPreferredHeight() {
        return getIntent().getIntExtra(CameraIntentKey.VIDEO_PREFERRED_HEIGHT, 720);
    }`

Max0n, 2016-06-07:
Строчку-то нашел. Поменял аспект на 16 / 9 и качество с 720 на 1080p. Но результата не дало. Видео зернистое и нет фокусировки.

sfilmak, 2016-06-07:
Попробуйте убрать из MainActivity все дополнительные настройки и вернуть все к начальному состоянию:
`new MaterialCamera(this)
                .saveDir(saveDir)
                .showPortraitWarning(true)
                .allowRetry(true)
                .defaultToFrontFacing(false)`
Если не поможет, то, увы, ничем больше помочь не могу. У меня на Xiaomi Mi3 с Android 6.0 (AOSP) изменение соотношения сторон решило проблему качества.

Max0n, 2016-06-07:
Ситуация стала буквально чуть-чуть лучше, но зернистость по сравнению со стоковой камерой все же присутствует и это сильно портит качество. Спасибо за совет, надеюсь разработчик обратит внимание на проблему и исправит ее.

closed 2016-09-25 by afollestad