afollestad/material-camera #64: Getting java.util.MissingFormatArgumentException After Clicking stop video record button.

pingothedoer, 2016-06-23:
`private void recordVideo(String fileName){`
	`File saveFoler = new File(fileName);`
	`if (!saveFoler.exists() || ! saveFoler.isDirectory()){`
		`saveFoler.mkdirs();`
	`}`

	new MaterialCamera(this)                               // Constructor takes an Activity
			.allowRetry(true)                                  // Whether or not 'Retry' is visible during playback
			.autoSubmit(false)                                 // Whether or not user is allowed to playback videos after recording. This can affect other things, discussed in the next section.
			.saveDir(saveFoler)                               // The folder recorded videos are saved to
			.primaryColorAttr(R.attr.colorPrimary)             // The theme color used for the camera, defaults to colorPrimary of Activity in the constructor
			.showPortraitWarning(true)                         // Whether or not a warning is displayed if the user presses record in portrait orientation
			.defaultToFrontFacing(false)                       // Whether or not the camera will initially show the front facing camera
			.retryExits(false)                                 // If true, the 'Retry' button in the playback screen will exit the camera instead of going back to the recorder
			.restartTimerOnRetry(false)                        // If true, the countdown timer is reset to 0 when the user taps 'Retry' in playback
			.continueTimerInPlayback(false)                    // If true, the countdown timer will continue to go down during playback, rather than pausing.
			.videoEncodingBitRate(1024000)                     // Sets a custom bit rate for video recording.
			.audioEncodingBitRate(50000)                       // Sets a custom bit rate for audio recording.
			.videoFrameRate(24)                                // Sets a custom frame rate (FPS) for video recording.
			.qualityProfile(MaterialCamera.QUALITY_HIGH)       // Sets a quality profile, manually setting bit rates or frame rates with other settings will overwrite individual quality profile settings
			.videoPreferredHeight(720)                         // Sets a preferred height for the recorded video output.
			.videoPreferredAspect(4f / 3f)                     // Sets a preferred aspect ratio for the recorded video output.
			.maxAllowedFileSize(1024 * 1024 * 5)               // Sets a max file size of 5MB, recording will stop if file reaches this limit. Keep in mind, the FAT file system has a file size limit of 4GB.
			.iconRecord(R.drawable.mcam_action_play)           // Sets a custom icon for the button used to start recording
			.iconStop(R.drawable.mcam_action_stop)             // Sets a custom icon for the button used to stop recording
			.iconFrontCamera(R.drawable.mcam_camera_front)     // Sets a custom icon for the button used to switch to the front camera
			.iconRearCamera(R.drawable.mcam_camera_rear)       // Sets a custom icon for the button used to switch to the rear camera
			.iconPlay(R.drawable.evp_action_play)              // Sets a custom icon used to start playback
			.iconPause(R.drawable.evp_action_pause)            // Sets a custom icon used to pause playback
			.iconRestart(R.drawable.evp_action_restart)        // Sets a custom icon used to restart playback
			.labelRetry(R.string.mcam_retry)                   // Sets a custom button label for the button used to retry recording, when available
			.labelUseVideo(R.string.mcam_use_video)            // Sets a custom button label for the button used to confirm a recording
			.start(CAMERA_RQ);
}

###### LogCat 
E/AndroidRuntime: FATAL EXCEPTION: main
				  Process: com.xyz.xyz, PID: 9203
				  java.util.MissingFormatArgumentException: Format specifier: 20v
					  at java.util.Formatter.getArgument(Formatter.java:1111)
					  at java.util.Formatter.doFormat(Formatter.java:1076)
					  at java.util.Formatter.format(Formatter.java:1042)
					  at java.util.Formatter.format(Formatter.java:1011)
					  at java.lang.String.format(String.java:1803)
					  at java.lang.String.format(String.java:1777)
					  at com.afollestad.easyvideoplayer.EasyVideoPlayer.LOG(EasyVideoPlayer.java:833)
					  at com.afollestad.easyvideoplayer.EasyVideoPlayer.prepare(EasyVideoPlayer.java:361)
					  at com.afollestad.easyvideoplayer.EasyVideoPlayer.onSurfaceTextureAvailable(EasyVideoPlayer.java:558)
					  at android.view.TextureView.getHardwareLayer(TextureView.java:370)
					  at android.view.View.updateDisplayListIfDirty(View.java:15373)
					  at android.view.View.getDisplayList(View.java:15418)
					  at android.view.View.draw(View.java:16188)
					  at android.view.ViewGroup.drawChild(ViewGroup.java:3709)
					  at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3502)
					  at android.view.View.draw(View.java:16469)
					  at android.widget.FrameLayout.draw(FrameLayout.java:598)
					  at android.view.View.updateDisplayListIfDirty(View.java:15396)
					  at android.view.View.getDisplayList(View.java:15418)
					  at android.view.View.draw(View.java:16188)
					  at android.view.ViewGroup.drawChild(ViewGroup.java:3709)
					  at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3502)
					  at android.view.View.draw(View.java:16469)
					  at android.widget.FrameLayout.draw(FrameLayout.java:598)
					  at android.view.View.updateDisplayListIfDirty(View.java:15396)
					  at android.view.View.getDisplayList(View.java:15418)
					  at android.view.View.draw(View.java:16188)
					  at android.view.ViewGroup.drawChild(ViewGroup.java:3709)
					  at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3502)
					  at android.view.View.updateDisplayListIfDirty(View.java:15391)
					  at android.view.View.getDisplayList(View.java:15418)
					  at android.view.View.draw(View.java:16188)
					  at android.view.ViewGroup.drawChild(ViewGroup.java:3709)
					  at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3502)
					  at android.view.View.updateDisplayListIfDirty(View.java:15391)
					  at android.view.View.getDisplayList(View.java:15418)
					  at android.view.View.draw(View.java:16188)
					  at android.view.ViewGroup.drawChild(ViewGroup.java:3709)
					  at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3502)
					  at android.view.View.updateDisplayListIfDirty(View.java:15391)
					  at android.view.View.getDisplayList(View.java:15418)
					  at android.view.View.draw(View.java:16188)
					  at android.view.ViewGroup.drawChild(ViewGroup.java:3709)
					  at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3502)
					  at android.view.View.updateDisplayListIfDirty(View.java:15391)
					  at android.view.View.getDisplayList(View.java:15418)
					  at android.view.View.draw(View.java:16188)
					  at android.view.ViewGroup.drawChild(ViewGroup.java:3709)
					  at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3502)
					  at android.view.View.draw(View.java:16469)
					  at android.widget.FrameLayout.draw(FrameLayout.java:598)
					  at com.android.internal.policy.impl.PhoneWindow$DecorView.draw(PhoneWindow.java:3107)
					  at android.view.View.updateDisplayListIfDirty(View.java:15396)
					  at android.view.View.getDisplayList(View.java:15418)
					  at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:310)
					  at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:316)
					  at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:355)
					  at android.view.ViewRootImpl.draw(ViewRootImpl.java:2890)
					  at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2684)
					  at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2271)
					  at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1264)
					  at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6943)
					  at android.view.Choreographer$CallbackRecord.run(Choreographer.java:777)
					  at android.view.Choreographer.doCallbacks(Choreographer.java:590)
					  at android.view.Choreographer.doFrame(Choreographer.java:560)
					at android.view.Choreographe

###### Include the following:
 - Material Camera version: `0.2.13`
 - Device OS version: `5.1`
 - Device Manufacturer: `Samsun`
 - Device Name: `Galaxy S6`

pingothedoer, 2016-06-23:
@afollestad can u plx take a look at it.

afollestad, 2016-06-23:
I get notified by email of new issues without you mentioned me just so you know :wink:

---

The log shows it's actually in [Easy Video Player](https://github.com/afollestad/easy-video-player), not this library. But the problem is related to the URI you're loading.

However, it should never happen if you're using the latest version of this library.

closed 2016-06-23 by afollestad

pingothedoer, 2016-06-24:
I am using the latest one, but if its URI related issue, at least it should not make my app crash. :-(

ibrahimGhailani, 2016-08-10:
I'm facing the same issue. Clicking the stop button when device is set to RTL language causes a FC.

```groovy 
E/AndroidRuntime: FATAL EXCEPTION: main
                                                                  Process: , PID: 5548
java.util.MissingFormatArgumentException: Format specifier: D
                                                                      at java.util.Formatter.getArgument(Formatter.java:1111)
                                                                      at java.util.Formatter.doFormat(Formatter.java:1076)
                                                                      at java.util.Formatter.format(Formatter.java:1042)
                                                                      at java.util.Formatter.format(Formatter.java:1011)
                                                                      at java.lang.String.format(String.java:1554)
                                                                      at java.lang.String.format(String.java:1528)
                                                                      at com.afollestad.easyvideoplayer.EasyVideoPlayer.LOG(EasyVideoPlayer.java:833)
                                                                      at com.afollestad.easyvideoplayer.EasyVideoPlayer.prepare(EasyVideoPlayer.java:361)
                                                                      at com.afollestad.easyvideoplayer.EasyVideoPlayer.onSurfaceTextureAvailable(EasyVideoPlayer.java:558)
                                                                      at android.view.TextureView.getHardwareLayer(TextureView.java:368)
                                                                      at android.view.View.updateDisplayListIfDirty(View.java:15157)
                                                                      at android.view.View.draw(View.java:15954)
                                                                      at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                      at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                      at android.view.View.draw(View.java:16187)
                                                                      at android.view.View.updateDisplayListIfDirty(View.java:15180)
                                                                      at android.view.View.draw(View.java:15954)
                                                                      at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                      at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                      at android.view.View.draw(View.java:16187)
                                                                      at android.view.View.updateDisplayListIfDirty(View.java:15180)
                                                                      at android.view.View.draw(View.java:15954)
                                                                      at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                      at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                      at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                      at android.view.View.draw(View.java:15954)
                                                                      at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                      at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                      at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                      at android.view.View.draw(View.java:15954)
                                                                      at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                      at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                      at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                      at android.view.View.draw(View.java:15954)
                                                                      at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                      at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                      at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                      at android.view.View.draw(View.java:15954)
                                                                      at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                      at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                      at android.view.View.draw(View.java:16187)
[... cut ...]
                                                                      at android.app.ActivityThread.main(ActivityThread.java:5422)
                                                                      at java.lang.reflect.Method.invoke(Native Method)
                                                                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
```
Material Camera: 0.2.14
LG Nexus 5X (6.0.1) and Moto G (5.1)

FaaaX, 2016-08-17:
I'm facing the same issue. Clicking the stop button. I'm using the newest version. 

`E/Report ::: java.util.MissingFormatArgumentException: Format specifier: 20S
                                                                 
                                                                 --------- Stack trace ---------
                                                                 
                                                                     at java.util.Formatter.getArgument(Formatter.java:1111)
                                                                     at java.util.Formatter.doFormat(Formatter.java:1076)
                                                                     at java.util.Formatter.format(Formatter.java:1042)
                                                                     at java.util.Formatter.format(Formatter.java:1011)
                                                                     at java.lang.String.format(String.java:1554)
                                                                     at java.lang.String.format(String.java:1528)
                                                                     at com.afollestad.easyvideoplayer.EasyVideoPlayer.LOG(EasyVideoPlayer.java:833)
                                                                     at com.afollestad.easyvideoplayer.EasyVideoPlayer.prepare(EasyVideoPlayer.java:361)
                                                                     at com.afollestad.easyvideoplayer.EasyVideoPlayer.onSurfaceTextureAvailable(EasyVideoPlayer.java:558)
                                                                     at android.view.TextureView.getHardwareLayer(TextureView.java:368)
                                                                     at android.view.View.updateDisplayListIfDirty(View.java:15173)
                                                                     at android.view.View.draw(View.java:15969)
                                                                     at android.view.ViewGroup.drawChild(ViewGroup.java:3612)
                                                                     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3402)
                                                                     at android.view.View.draw(View.java:16202)
                                                                     at android.view.View.updateDisplayListIfDirty(View.java:15196)
                                                                     at android.view.View.draw(View.java:15969)
                                                                     at android.view.ViewGroup.drawChild(ViewGroup.java:3612)
                                                                     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3402)
                                                                     at android.view.View.draw(View.java:16202)
                                                                     at android.view.View.updateDisplayListIfDirty(View.java:15196)
                                                                     at android.view.View.draw(View.java:15969)
                                                                     at android.view.ViewGroup.drawChild(ViewGroup.java:3612)
                                                                     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3402)
                                                                     at android.view.View.updateDisplayListIfDirty(View.java:15191)
                                                                     at android.view.View.draw(View.java:15969)
                                                                     at android.view.ViewGroup.drawChild(ViewGroup.java:3612)
                                                                     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3402)
                                                                     at android.view.View.updateDisplayListIfDirty(View.java:15191)
                                                                     at android.view.View.draw(View.java:15969)
                                                                     at android.view.ViewGroup.drawChild(ViewGroup.java:3612)
                                                                     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3402)
                                                                     at android.view.View.updateDisplayListIfDirty(View.java:15191)
                                                                     at android.view.View.draw(View.java:15969)
                                                                     at android.view.ViewGroup.drawChild(ViewGroup.java:3612)
                                                                     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3402)
                                                                     at android.view.View.updateDisplayListIfDirty(View.java:15191)
                                                                     at android.view.View.draw(View.java:15969)
                                                                     at android.view.ViewGroup.drawChild(ViewGroup.java:3612)
                                                                     at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3402)
[... cut ...]
                                                                     at android.app.ActivityThread.main(ActivityThread.java:5443)
                                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:728)
                                                                     at com.android.internal.os.ZygoteInit.main(ZygoteIn`

FaaaX, 2016-08-22:
Fixed in 0.3.0.