afollestad/material-camera #71: Output file isn't always an mp4 file

dabele91, 2016-07-12:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.2.14`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `Huawei`
 - Device Name: `Nexus 6P`

Also, please wrap Java with correct syntax highlighting.

```java
new MaterialCamera(fragment.getActivity())
                        .allowRetry(true)
                        .autoSubmit(false)
                        .saveDir(fragment.getContext().getExternalCacheDir())
                        .primaryColorAttr(R.attr.colorPrimary)
                        .showPortraitWarning(false)
                        .maxAllowedFileSize(1024 * 1024 * 20)
                        .qualityProfile(MaterialCamera.QUALITY_LOW)
                        .start(CAMERA_RQ);
```
 
###### Reproduction Steps

1. Use the following code
2. check the saved mp4 file. it's an 3gp file not an mp4 file.

###### Expected Result

Because the file ending is mp4, the expected result is an mp4 file.

###### Actual Result

The file is saved with an extension mp4, but it's an 3gp file.

afollestad, 2016-07-12:
Hmm. Well it might be changed if you set quality profiles. Some devices might fall back to 3GP if they don't support MP4, too.

dabele91, 2016-07-12:
Yes i know. The output format is mp4 if i'm changing the quality to 480p or something else. But in my opinion, if somebody uses the quality profile low or the default camera output format for the given quality level isn't mp4, the file should be saved with the proper file extension.
If the file extension is right, the user of your library is able to check the mime type of the extension and can do some conversion stuff if he needs an mp4 file.

afollestad, 2016-08-20:
As far as I know, there's not a good way to know for sure what the output format will be. 

I'm pretty surprised any modern device would use 3GP, the last time I had a phone that recorded 3GP videos, it was a flip phone.

closed 2016-11-07 by afollestad