afollestad/material-camera #73: Error:Failed to resolve: com.afollestad:easyvideoplayer:0.2.8

PaoloRotolo, 2016-07-23:
I'm unable to use latest Material Camera in my project for a missing dependency.
Please fix ASAP.

The missing dependency is: ```com.afollestad:easyvideoplayer:0.2.8``` I can't find it upstream.

PaoloRotolo, 2016-07-23:
Issue #59 was closed, but the problem still persists.

kazuooooo, 2016-07-24:
I also has the same issue...

kazuooooo, 2016-07-24:
passed add this to build.grale like #56 
```
repositories {
    maven { url "https://dl.bintray.com/drummer-aidan/maven" }
}
```

closed 2016-07-24 by afollestad

gokceyucel, 2016-07-28:
There 's still a bug in this final release. material-camera 0.2.14 depends on easy-video-player 0.2.8 but there's a bug in it that I experienced.

Here 's the related bug, it is not fixed in 0.2.9 but in master branch.

https://github.com/afollestad/easy-video-player/blob/0.2.8/library/src/main/java/com/afollestad/easyvideoplayer/EasyVideoPlayer.java#L831

https://github.com/afollestad/easy-video-player/blob/master/library/src/main/java/com/afollestad/easyvideoplayer/EasyVideoPlayer.java#L859

I think material-camera library should depend on upcoming release (currently master).
Application crashes while using easy-video-player 0.2.9 with material-camera 0.2.14