afollestad/material-camera #83: Surfaceview preview stretching

msr550, 2016-08-24:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.3.0`
 - Device OS version: `6.0`
 - Device Manufacturer: `Lenovo`
 - Device Name: `Vibe X3`

 - Material Camera version: `0.3.0`
 - Device OS version: `5.1`
 - Device Manufacturer: `Redmi`
 - Device Name: `Prime2`

 
###### Reproduction Steps

1. Integrated libarary 0.3.0
2. Open Camera Preview
3. Camera preview looks stretched

###### Expected Result
1. Camera preview should support (best preview) for all the devices.

###### Actual Result
1. Camera previews are not supporting looks stretch, after captured photo its looking good.

donjerick, 2016-08-29:
Stretched camera preview also on Samsung Galaxy Note 5 using library version 0.3.0

donjerick, 2016-08-29:
Managed to solve this by adding:

wscheffer, 2016-09-05:
When the aspect ratio is set to 4/3 the SurfaceView is still stretched to fit the screen which causes the image to stretch

XanthusL, 2016-09-07:
The SurfaceView fits the screen size. So it's stretched

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/issues/90

closed 2016-09-25 by afollestad