afollestad/material-camera #84: Playback and countdownSeconds

BackPackerDz, 2016-08-27:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.3.0`
 - Device OS version: `4.2`
 - Device Manufacturer: `Huawei`
 - Device Name: `Huawei P6`

Also, please wrap Java with correct syntax highlighting.

```java
  MaterialCamera materialCamera = new MaterialCamera(this)
                .saveDir(saveDir)
                .showPortraitWarning(false)
                .countdownSeconds(2)
                .allowRetry(true)
                .autoSubmit(false)
                .defaultToFrontFacing(true)
```
###### Expected Result
When I use ".countdownSeconds", the demo doesnt' send me to the playback part !

UglyTroLL, 2016-09-06:
+1 Same issue here

aelimill, 2016-09-16:
same issue

closed 2016-09-25 by afollestad

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.0

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.1

commit 69565c94 (2016-09-25): Resolved #84