afollestad/material-camera #85: Stillshot Activity doesn't have any way of setting UseVideo button's Text

wscheffer, 2016-08-31:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.3.2`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `HTC`
 - Device Name: `HTC 10`

Also, please wrap Java with correct syntax highlighting.

```java
 MaterialCamera materialCamera = new MaterialCamera(this)
                       .stillShot()
                       .labelUseVideo(R.string.use_image)
```

###### Expected Result
On Stillshot Activity "mUseVideo" button text should show Resource string "use_image" string value

###### Actual Result
Shows text of resource R.strings.mcam_use_stillshot

wscheffer, 2016-08-31:
I have a fix for this issue but don't have access to push the branch

mowcixo, 2016-08-31:
@wscheffer Yo mind that it should be another method called ```labelUseStillshot``` to change the label's text in stillshot mode?

wscheffer, 2016-08-31:
My first assumption was that .labelUseVideo(R.string.ok) would work but I don't see any code in StillshotPreviewFragment for mUseVideo.setText(mInterface.labelUseVideo()) ?

mowcixo, 2016-08-31:
@wscheffer yes, you're right, you can create a pull request to propose your changes by following this guide: https://help.github.com/articles/creating-a-pull-request/

wscheffer, 2016-08-31:
Hey, creating the PR is not an issues problem comes in when I try to push the branch I receive the below error i'm not sure if i'm missing something

--------------------------------------------------
"origin" refs/heads/feature/stillshot_labels:refs/heads/feature/stillshot_labels
ERROR: Permission to afollestad/material-camera.git denied to wscheffer.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
--------------------------------------------------

mowcixo, 2016-08-31:
@wscheffer oh, of course! you need to fork afollestad's repo, to one of your own, make the changes on it, and then create the pull request over Aidan's original repo... so he can accept the modification or review it.

wscheffer, 2016-08-31:
Ok done, thank you for all the help

mowcixo, 2016-08-31:
@wscheffer You're welcome. So the solution of this issue is proposed in #86.

closed 2016-09-02 by afollestad

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.0

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.1