afollestad/material-camera #90: Video Preview Looks Stretched on Nexus 5

f2dude, 2016-09-02:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.3.2`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `LG`
 - Device Name: `Nexus 5`

I used the following code to video preview on Nexus 5:

```java
materialCamera.videoPreferredAspect(16f/9f); //Video preview doesn't stretch on Xiomi Redmi and Lenovo Vibe x3 but a blank space appears on the right side of video preview for Nexus 5
materialCamera.videoPreferredAspect(4f/3f); //Video preview doesn't stretch on Nexus 5 but looks stretched on other devices such as Lenovo Vibe x3 and Xiomi Redmi
```
 
###### Reproduction Steps

1. 
2. 
3. 

###### Expected Result
I know this is a difficult question but is it possible to make the video preview look perfectly on all the devices
[image]

###### Actual Result
[image]

afollestad, 2016-09-26:
As I'm sure you noticed, it's hard to get it to work on all devices ;-)

mdolnik, 2016-10-08:
I don't have this issue on my Nexus 5, but I can confirm this is an issue on my other test devices:

- Material Camera version: 0.4.2
- Device OS version: 6.0.1 / 4.0.4 / 4.1.2
- Device Manufacturer: LG / Sony / Unknown
- Device Name: Nexus 5 / Xperia LT15i / ET-Q8

One quick way to test whether an image is stretched is to film a sphere of some sort (ie: tennis ball) that way no matter what angle you look at, the result should be a perfect circle.

Adding a bit of xml to the end of the layout file: _mcam_fragment_videocapture.xml_ file(s) can overlay a simple perfect square to compare against a sphere.

```
<View
        android:id="@+id/colored_bar"
        android:layout_width="200dp"
        android:background="@android:color/holo_green_dark"
        android:layout_centerVertical="true"
        android:layout_centerHorizontal="true"
        android:alpha="0.5"
        android:layout_height="200dp" />
```

**Sony Xperia** (stretched)
[image]

**Sony Xperia** (stretched)
[image]

**ET-Q8** (stretched)
[image]

**ET-Q8** (stretched)
[image]

**Nexus 5** (seems fine)
[image]

**Nexus 5** (seems fine)
[image]

mdolnik, 2016-11-12:
So I think I've narrowed down where the issue is for the two older devices that use the older camera API: `CameraFragment.java`

In general it seems to assume that the camera preview will always be the same aspect ratio as the display/device, correct me if I'm wrong.

In `createPreview()` it sets the preview's aspect ratio to the width/height of the activity via         `mPreviewView.setAspectRatio(mWindowSize.x, mWindowSize.y);`

For both of my devices with the issues, it turned out the stretching was happening because the largest capture/preview size supported by their cameras were 3:2 and the screen sizes are 16:9

**I found the following code change in `createPreview()` fixes this issue for my two devices:**
Remove these lines from the top:
```
Activity activity = getActivity();
if (activity == null) return;
if (mWindowSize == null)
    mWindowSize = new Point();
activity.getWindowManager().getDefaultDisplay().getSize(mWindowSize);
```
Add these lines to the bottom (and replace the setAspectRatio line):
```
// Find the camera's preview size and current rotation
Camera.Size previewSize = mCamera.getParameters().getPreviewSize();
int aspectWidth = previewSize.width;
int aspectHeight = previewSize.height;
// If the rotation is 90 or 270, flip the aspect ratio values
int rotation = Integer.parseInt(mCamera.getParameters().get("rotation"));
if (rotation == Degrees.DEGREES_90 || rotation == Degrees.DEGREES_270) {
    aspectWidth = previewSize.height;
    aspectHeight = previewSize.width;
}
mPreviewView.setAspectRatio(aspectWidth, aspectHeight);
```

With all of that said this issue only happened for me with the devices using the older camera APIs and not my nexus 5. The code change above won't affect the camera2 api so it wouldn't help any issues with op.

mdolnik, 2017-03-03:
Has any work been done on this? Should I submit a pull request with my fix for camera 1 api?

aftabsikander, 2017-03-06:
@mdolnik Its not fixed, I have seen this issue as well It would be great if you submit pull request for this.

ejezisek, 2017-04-08:
I did it slightly differently as per: http://stackoverflow.com/questions/19577299/android-camera-preview-stretched

This was needed because while your fix above fixed it on several phones it seemed like the issue was still present on a Samsung Galaxy S5.  I did not see this issue present on any device that uses Camera2 including a Google pixel and a Nexus 5.

I made some changes to support the rotation if a video is being taken.  Please let me know if anyone has any issues with the pull request I made.  Currently it is failing due to using Android Build tools 25 in the compilation.

BabuinGH, 2017-06-09:
I still see on Samsung S5 that the video preview is stretched. I just need to capture portrait, show the preview on 1:1 ratio and center vertically. Any suggestions ?

ejezisek, 2017-06-09:
Babuin, my pull request has not been merged.  Feel free to download it and see if it works for you.  I'm pretty sure I specifically tested the S5 on that pull request.

akash-pal, 2017-07-26:
@ejezisek please raise another pull request. 
It works fine on redmi note 3 ; originally it was getting stretched.

bruferrari, 2017-08-06:
this issue have been solved? I'm still getting streched image on Moto G5 in portrait or landscape

ejezisek, 2017-08-06:
@akash-pal my pull request is still open.  Are you suggesting some changes to my pull request?  I haven't looked at this for a while.

bruferrari, 2017-08-07:
@ejezisek did you find a way to correct the aspect ratio? I mean, fix the stretched preview... I've been forked this lib in order to use in a professional project and since I've forked and managed some mods I'm looking for a way to correct this issue

ejezisek, 2017-08-07:
@bruferrari I would recommend going to the open pull request section and downloading my pull request.  It has not been merged into the main library as of this point, but per my tests it was successful (however they were by no means exhaustive.). Please let me know if you have any issues.