afollestad/material-camera #91: getting "CameraFragment: Couldn't find any suitable preview size" on samsung s4 lollypop

shafqatjamil, 2016-09-05:
09-05 11:31:30.901 22651-22651/com.liquidsoftwaremobile.mytarotadvisor I/Timeline: Timeline: Activity_launch_request id:com.liquidsoftwaremobile.mytarotadvisor time:24487234
09-05 11:31:30.916 22651-22982/com.liquidsoftwaremobile.mytarotadvisor V/FA: Activity paused, time: 92577779
09-05 11:31:30.981 22651-22651/com.liquidsoftwaremobile.mytarotadvisor V/FA: onActivityCreated
09-05 11:31:31.046 22651-22651/com.liquidsoftwaremobile.mytarotadvisor D/Activity: performCreate Call secproduct feature valuefalse
09-05 11:31:31.046 22651-22651/com.liquidsoftwaremobile.mytarotadvisor D/Activity: performCreate Call debug elastic valuetrue
09-05 11:31:31.071 22651-22982/com.liquidsoftwaremobile.mytarotadvisor V/FA: Activity resumed, time: 92577939
09-05 11:31:31.151 22651-22651/com.liquidsoftwaremobile.mytarotadvisor D/CameraFragment: Couldn't find any suitable preview size
09-05 11:31:31.151 22651-22651/com.liquidsoftwaremobile.mytarotadvisor D/stillshot: using resolution: 1920x1080
09-05 11:31:31.156 22651-22651/com.liquidsoftwaremobile.mytarotadvisor D/CameraFragment: Orientations: Sensor = 270˚, Device = 0˚, Display = 270˚
09-05 11:31:31.306 22651-22982/com.liquidsoftwaremobile.mytarotadvisor V/FA: Activity paused, time: 92578173
09-05 11:31:31.366 22651-22982/com.liquidsoftwaremobile.mytarotadvisor V/FA: Activity resumed, time: 92578234
Please help me

ruha9005, 2016-09-13:
Sony XPeria SP, Android 4.3.
Catching this error when `.videoPreferredHeight(1080)`

tylerkrupicka, 2016-10-15:
I got my hands on a Galaxy S4 running 5.0.1 today and was unable to reproduce this issue in material-camera 0.4.2. I tried all of the fragments, stillshot, and video and had no issues with the current build with the https://github.com/sugarmanz/material-camera/ patches.

closed 2016-10-18 by afollestad