afollestad/material-camera #92: Calling MaterialCamera().stillShot().start() on Android 4.3 blocks on permission check

teeeeeegz, 2016-09-06:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.3.2`
 - Device OS version: `4.3`
 - Device Manufacturer: `Genymotion`
 - Device Name: `Genymotion - Samsung Galaxy S3 settings`

 
###### Reproduction Steps

1. Execute this block of code from a Fragment, using Support v4 Library
```java
    void avatarLayoutClicked() {
        Timber.i("Avatar Layout Clicked");
        new MaterialCamera(getActivity())
                .stillShot()
                .start(CAMERA_REQUEST);
    }
```

2. Click on layout to execute the above code

###### Expected Result
- Logcat to output: "Avatar Layout Clicked"
- Camera capture activity to open

###### Actual Result
- Logcat outputs "I/SignupFragment:121: Avatar Layout Clicked"
- Logcat returns these two errors:
-- Could not find method android.view.Window.setStatusBarColor, referenced from method com.afollestad.materialcamera.internal.BaseCaptureActivity.onCreate
-- Could not find method android.app.Activity.checkSelfPermission, referenced from method com.afollestad.materialcamera.internal.CameraFragment.prepareMediaRecorder

I looked at the source code and this code is called
```java
private boolean prepareMediaRecorder() {
        try {
            final Activity activity = getActivity();
            if (null == activity) return false;
            final BaseCaptureInterface captureInterface = (BaseCaptureInterface) activity;

            setCameraDisplayOrientation(mCamera.getParameters());
            mMediaRecorder = new MediaRecorder();
            mCamera.stopPreview();
            mCamera.unlock();
            mMediaRecorder.setCamera(mCamera);

            boolean canUseAudio = true;
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M)
                canUseAudio = activity.checkSelfPermission(Manifest.permission.RECORD_AUDIO) == PackageManager.PERMISSION_GRANTED;

            if (canUseAudio) {
                mMediaRecorder.setAudioSource(MediaRecorder.AudioSource.DEFAULT);
            } else {
                Toast.makeText(getActivity(), R.string.mcam_no_audio_access, Toast.LENGTH_LONG).show();
            }
}
}
```

...but the Permission check which references the activity object, should only occur if the Build.VERSION.SDK_INT is Marshmallow or above.

Thoughts?

afollestad, 2016-09-25:
That code above is for video, not stillshot, and it works fine.

afollestad, 2016-09-25:
There's no way the methods referenced above could cause issues. They're only called on the API levels they are supported on.

If you are getting errors or warnings, you're probably using an old target SDK, an old version of Build Tools, or an old version of Android Studio.

closed 2016-09-25 by afollestad