afollestad/material-camera #93: How can I record a wide video when the orientation is portrait

XanthusL, 2016-09-06:
I need to record wide videos when the orientation is portrait. 
I have no idea to preview in part of the screen.

Tried to change the code in library, failed then. 
Is there anybody who can give me advice plz

franklinexpress, 2016-09-10:
Have you tried this

```
  .videoPreferredHeight(720)                         // Sets a preferred height for the recorded video output.
    .videoPreferredAspect(4f / 3f)                     // Sets a preferred aspect ratio for the recorded video output.
```

XanthusL, 2016-09-10:
@franklinexpress  I tried, but the video is 720 height and 540 width when the orientation is portrait.

afollestad, 2016-09-25:
Messing with configuration like the above is the only solution.

And this library can only support whatever your device allows and reports to Android's camera APIs.

closed 2016-09-25 by afollestad