afollestad/material-camera #98: Can't be use labelConfirm() method.

thitoo, 2016-09-12:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.2.13`
 - Device OS version: `7.0`
 - Device Manufacturer: `Huawei`
 - Device Name: `Nexus 6P`

Also, please wrap Java with correct syntax highlighting.

```java
System.out.println("Hello, world!");
```
 
###### Reproduction Steps

1. 
2. 
3. 

###### Expected Result

###### Actual Result

mowcixo, 2016-09-12:
If you're using version 0.2.13 is normal you can't, that's implemented in the master branch.

closed 2016-09-12 by afollestad

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.0

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.1