afollestad/material-camera #100: Stillshots cause out of memory exception

aguragorn, 2016-09-15:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.3.2`
 - Device OS version: `4.4.2`
 - Device Manufacturer: `Samsung`
 - Device Name: `Galaxy Note 2`

Every time I use the camera for still shots my app's memory usage significantly increases. After a couple shots, it crashes with out of memory exception.

Here is how I launch the camera.

```java
new MaterialCamera(this).stillShot()
    .saveDir(getPrivateDataDirectory())
    .start(ACTIVITY_CAPTURE_IMAGE);
```
getPrivateDataDirectory returns a directory from the app's private data directory

Here is the exception
```
java.lang.OutOfMemoryError
  at android.graphics.BitmapFactory.nativeDecodeStream(Native Method)
  at android.graphics.BitmapFactory.decodeStreamInternal(BitmapFactory.java:709)
  at android.graphics.BitmapFactory.decodeStream(BitmapFactory.java:685)
  at android.graphics.BitmapFactory.decodeFile(BitmapFactory.java:447)
  at com.afollestad.materialcamera.util.ImageUtil.getRotatedBitmap(ImageUtil.java:84)
  at com.afollestad.materialcamera.util.ImageUtil.getRotatedBitmap(ImageUtil.java:96)
  at com.afollestad.materialcamera.util.ImageUtil.getRotatedBitmap(ImageUtil.java:96)
  at com.afollestad.materialcamera.util.ImageUtil.getRotatedBitmap(ImageUtil.java:64)
  at com.afollestad.materialcamera.internal.StillshotPreviewFragment.setImageBitmap(StillshotPreviewFragment.java:74)
  at com.afollestad.materialcamera.internal.StillshotPreviewFragment.access$000(StillshotPreviewFragment.java:19)
  at com.afollestad.materialcamera.internal.StillshotPreviewFragment$1.onPreDraw(StillshotPreviewFragment.java:57)
  at android.view.ViewTreeObserver.dispatchOnPreDraw(ViewTreeObserver.java:888)
  at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2199)
  at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1254)
  at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6637)
  at android.view.Choreographer$CallbackRecord.run(Choreographer.java:814)
  at android.view.Choreographer.doCallbacks(Choreographer.java:614)
  at android.view.Choreographer.doFrame(Choreographer.java:584)
  at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:800)
  at android.os.Handler.handleCallback(Handler.java:733)
  at android.os.Handler.dispatchMessage(Handler.java:95)
  at android.os.Looper.loop(Looper.java:146)
  at android.app.ActivityThread.main(ActivityThread.java:5602)
  at java.lang.reflect.Method.invokeNative(Native Method)
  at java.lang.reflect.Method.invoke(Method.java:515)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1283)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1099)
  at dalvik.system.NativeStart.main(Native Method)
```

sugarmanz, 2016-09-15:
Thanks for the info. There is currently a patch being written that
addresses this issue. It only seems to affect Samsung devices.

afollestad, 2016-09-26:
@sugarmanz really? What's the source of the problem?

JehandadK, 2016-10-11:
Any proposed workaround for now?

JehandadK, 2016-10-12:
```
 public static Bitmap getRotatedBitmap(String inputFile, int reqWidth, int reqHeight) {
        int IN_SAMPLE_SIZE_DEFAULT_VAL = 1;
        if (reqHeight > 2000) IN_SAMPLE_SIZE_DEFAULT_VAL = 4;
        else if (reqHeight > 1000) IN_SAMPLE_SIZE_DEFAULT_VAL = 2;
        return getRotatedBitmap(inputFile, reqWidth, reqHeight, IN_SAMPLE_SIZE_DEFAULT_VAL);
    }
```

tylerkrupicka, 2016-10-15:
@sugarmanz and I have a fix for this in https://github.com/sugarmanz/material-camera/commit/f6db69aaa0503353c17e6a58f6e9a799c46ccecc. The issue is with the calculateSampleSize logic, which was not downsampling enough for large images (which is extremely apparent on Samsung devices). We will submit this as part of a much larger patch soon, after we complete a round of testing across all of our devices. We now have access to pretty much every Samsung device, so we will pay extra attention to memory usage on those.

closed 2016-10-18 by afollestad

ketanbhardwaj, 2018-01-09:
Have you guys handled this out of memory issue in v0.4.4 ?
Because I am still getting this same error