afollestad/material-camera #101: Please add onRecordingStarted() and on RecordingFinished() to run custom code

PaoloRotolo, 2016-09-15:
- [ x ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ x ] I have verified that I am using the latest version of Material Camera.
- [ x ] I have given my issue a non-generic title.

Please include some methods like ```onRecordingStarted()``` and ```onRecordingFinished``` that we can override in order to run custom code during the video recording.

For example, I need to do some facial recognition while recording the video and show the output on the camera preview. So I need to run my code as soon as the user taps on the Recording button.

Thank you.

afollestad, 2016-09-25:
That can't really be done. Callbacks are *not* maintained through orientation change, *and* the camera is running in a `Fragment` within a separate `Activity`.

afollestad, 2016-09-25:
For you, the best solution would be to fork this library and add the facial recognition code into it.

closed 2016-09-25 by afollestad