afollestad/material-camera #99: onPictureTaken not called on Samsung Ace

andhie, 2016-09-15:
- [✓] I have verified there are no duplicate active or recent bugs, questions, or requests
- [✓ ] I have verified that I am using the latest version of Material Camera.
- [✓ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: Latest master [9897a12ffd60e55cc1cf502e460e6dfd5afe82de]
 - Device OS version: 4.4.4 (stock ROM)
 - Device Manufacturer: Samsung
 - Device Name: Samsung Galaxy Ace 4 (SM-G357FZ)
 
###### Reproduction Steps

1.  Clone the Master branch and compile the sample
2. Select "Launch Camera for Stillshot"
3. Switch to Back Camera (its defaulted to Front in the code)
4. Click on the Camera icon to take picture
5. onPictureTaken() will never get triggered and camera button will never be enabled again
https://github.com/afollestad/material-camera/blob/9897a12ffd60e55cc1cf502e460e6dfd5afe82de/library/src/main/java/com/afollestad/materialcamera/internal/CameraFragment.java#L515

###### Expected Result
Picture is taken

###### Actual Result
onPictureTaken never get called

###### Other observations
- When using Front camera, it was able to get callback
- The system logs clearly stated the camera successfully took a picture
```java
09-15 11:47:11.400 776-1052/? I/InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=109498512968000
09-15 11:47:11.400 776-1051/? I/InputDispatcher: Delivering touch to (29082): action: 0x1, toolType: 1
09-15 11:47:11.400 229-1937/? D/SecCameraCoreManager: enableMsgType: 0x302
09-15 11:47:11.400 229-1937/? W/QCamera2HWI: enable_msg_type : E, msg type 770
09-15 11:47:11.400 229-1937/? W/QCamera2HWI: enable_msg_type : X
09-15 11:47:11.400 229-1937/? D/SecCameraCoreManager: getParameters
09-15 11:47:11.400 229-1937/? W/QCamera2HWI: get_parameters : E
09-15 11:47:11.400 229-1937/? W/QCamera2HWI: get_parameters : X
09-15 11:47:11.400 229-1937/? W/QCamera2HWI: put_parameters : E
09-15 11:47:11.400 229-1937/? W/QCamera2HWI: put_parameters : X
09-15 11:47:11.400 229-1937/? D/SecCameraCoreManager: takePicture
09-15 11:47:11.400 776-1351/? I/EDMNativeHelperService: isCameraEnabled
09-15 11:47:11.400 229-1937/? D/SecCameraCoreManager: virtual android::status_t android::SecCameraCoreManager::takePicture():stop IT Policy checking thread
09-15 11:47:11.430 229-1937/? W/QCamera2HWI: [KPI Perf] send_command: E
09-15 11:47:11.430 229-31100/? E/QCamera2HWI: send command : 1509
09-15 11:47:11.430 229-1937/? W/QCamera2HWI: [KPI Perf] send_command: X
09-15 11:47:11.430 229-1937/? I/ShotSingle: ShotSingle::takePicture start
09-15 11:47:11.430 229-1937/? V/ShotSingle: takePicture(0)
09-15 11:47:11.430 229-1937/? E/QCamera2HWI: take_picture: af waiting count[0]
09-15 11:47:11.430 229-1937/? W/QCamera2HWI: take_picture
09-15 11:47:11.430 229-1937/? I/ShotSingle: ShotSingle::takePicture end
09-15 11:47:11.430 229-31213/? W/QCamera2HWI: [KPI Perf] take_picture_internal: E
09-15 11:47:11.430 229-31213/? W/QCamera2HWI: take_picture_internal Flash Mode=0, LLS mode=0, Auto LLS mode=0, NumOfSnaps=1
09-15 11:47:11.430 229-31100/? W/QCamera2HWI: takeLiveSnapshot: E mCameraId=0
09-15 11:47:11.430 229-31100/? E/QCamera2HWI: bool 
```
- Compiling the latest sample from Google/cameraview, it was able to get the onPictureTaken callback https://github.com/google/cameraview/blob/88c1eca7c4ef566e74598f3bee7ae3c81f8aacc8/library/src/main/camera1/com/google/android/cameraview/Camera1.java#L243

sugarmanz, 2016-10-21:
Hey @andhie, what is the string value of Build.DEVICE for your device?

andhie, 2016-10-26:
Here are some extra info for you

Build | Value 
-----|------
ID | KTU84P
DISPLAY | KTU84P.G357FZDXU1AOE3
PRODUCT | heatqltexx
DEVICE | heatqlte
BOARD | MSM8916
MANUFACTURER | samsung
BRAND | samsung
MODEL | SM-G357FZ
HARDWARE | qcom
CPU_ABI | armeabi-v7a
CPU_ABI2 | armeabi
TAGS | release-keys

tylerkrupicka, 2016-10-29:
Thanks for the information @andhie. We think this could be the same bug we saw with the Samsung Galaxy S3, which has been fixed but only for the GS3 device ID. @sugarmanz will create a branch with the patch applied for this device ID sometime soon and let you know.

afollestad, 2016-11-06:
https://github.com/afollestad/material-camera/releases/tag/0.4.3

closed 2016-11-06 by afollestad