afollestad/material-camera #102: Flashbutton will appear after switch back/front firstly

fegl84, 2016-09-16:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.3.2`
 - Device OS version: `6.0`
 - Device Manufacturer: `LG`
 - Device Name: `Nexus 5X`
 
My method to use material-camera:
`    public static void doUseCamera(Activity activity) {
                new MaterialCamera(activity)
                        /** all the previous methods can be called, but video ones would be ignored */
                        .stillShot() // launches the Camera in stillshot mode,
                        .primaryColorAttr(R.attr.colorPrimary)
                        .defaultToFrontFacing(false)
                        .saveDir(Config.CAMERA)
                        .start(SELECTED_CAMERA);

    }`

###### Reproduction Steps
1. start a camera with doUseCamera()

###### Expected Result
1. camera starts
2. Icon to Switch front / back appears bottom left
3. Icon to set flashmode apperas bottom left

###### Actual Result
3. Icon to set flashmode don't appear
4. after Switch to front and to back, a flashmode appears!

sugarmanz, 2016-09-16:
>    1. Icon to set flashmode don't appear
>    2. after Switch to front and to back, a flashmode appears!

A patch is being written that fixes this issue. Thanks for the info. FYI,
the flash button is actually still there. And clicking the button will
toggle the states, triggering logic to choose a drawable for the button.
This is a much quicker way to get the button to show up than  switching
between front-facing and rear camera. Again, this is just FYI to help make
your experience a little easier until the patch is done.

closed 2016-09-25 by afollestad

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.0

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.1

commit 8a83ce37 (2016-09-25): Fixed flash button not initially appearing when defaulting to rear facing camera (before switching front/back facing setting). Resolves #102