afollestad/material-camera #103: Actual image is upside down for stillshot

patelkarn, 2016-09-16:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.3.2`
 - Device OS version: `7.0` `6.0.1`
 - Device Manufacturer: `Huawei` `LG`
 - Device Name: `Nexus 6P` `Nexus 5X`
 
###### Reproduction Steps

1. Run sample Material Camera app
2. Hold device in landscape
3. Press LAUNCH CAMERA FOR STILLSHOT
4. Take picture
5. The final image will be upside down.

tylerkrupicka, 2016-09-19:
I can reproduce this on my Nexus 6p as well. I'll look in to this if I have time.

afollestad, 2016-09-25:
Unable to reproduce on my Nexus 6P with Android 7.0.

tylerkrupicka, 2016-09-25:
I'll check on my 6P again now that I'm on 7.0 as well.

afollestad, 2016-09-26:
Have you gotten the chance to check yet?

afollestad, 2016-09-26:
Closing for now since I haven't had issues on my 6P, or any other reports.

closed 2016-09-26 by afollestad

tylerkrupicka, 2016-09-27:
@afollestad: I cannot reproduce this issue on my Nexus 6p or my roommates on 0.4.1.

pschill, 2016-09-28:
It occurs if you disable auto rotation.

mdolnik, 2016-10-09:
It's not an issue on my Nexus 5 or Sony Xperia but this occurs for me for one of my devices, ET-Q8 (Unknown Manufacturer), I'm not sure what could be causing it on the one device.

btw a ET-Q8 is a crappy no-name tablet I got for free.

To clarify this may be a slightly different issue. For me, the camera preview for the front camera is upside down, and the result image screen / file has the correct orientation