afollestad/material-camera #105: Setting labelRetry and labelUseVideo do not change the value of the labels

thierrykoppenol, 2016-09-22:
- [X] I have verified there are no duplicate active or recent bugs, questions, or requests
- [X] I have verified that I am using the latest version of Material Camera.
- [X] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.3.2`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `Samsung`
 - Device Name: `Galaxy S5 (SM-G900F)`
 
###### Reproduction Steps

1.  Setting .labelRetry("someString") and .labelUseVideo("someString") in code
2. Starting the camera intent and taking a picture / video
3. Check the label below the taken picture / video

[image]

###### Expected Result
Labels displaying the strings as set in the methods

###### Actual Result
Labels are showing the default values (Retry and Use Video / Use Stillshot)

[image]

mowcixo, 2016-09-22:
I guess that's because the 0.3.2 is previous to this changes... try to use the master branch until there's a new tag.

thierrykoppenol, 2016-09-22:
Thanks, this has fixed the problem!

closed 2016-09-22 by thierrykoppenol

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.0

afollestad, 2016-09-25:
https://github.com/afollestad/material-camera/releases/tag/0.4.1