afollestad/material-camera #106: Error: Failed to resolve: com.afollestad:easyvideoplayer:0.2.13

DenisZhuravlev, 2016-09-26:
`compile 'com.afollestad:material-camera:0.4.1'`

causing  `Error:Failed to resolve: com.afollestad:easyvideoplayer:0.2.13`

Adding `maven { url "https://dl.bintray.com/drummer-aidan/maven" }` doesn't help

mowcixo, 2016-09-26:
Yep, that's probably true @afollestad, there is not published 0.2.13 version in bintray https://dl.bintray.com/drummer-aidan/maven/com/afollestad/easyvideoplayer/

Temporarily you can add the lib this way, @DenisZhuravlev:

```gradle

    compile 'com.afollestad:easyvideoplayer:0.2.12'
    compile ('com.afollestad:material-camera:0.4.1') {
        exclude group: 'com.afollestad', module: 'easyvideoplayer'
    }
```

afollestad, 2016-09-26:
Woops, it's published now.

https://bintray.com/drummer-aidan/maven/easy-video-player/0.2.13

closed 2016-09-26 by afollestad

mowcixo, 2016-09-26:
Perfect. It should work now without changes @DenisZhuravlev.