afollestad/material-camera #109: Still Shot Crashes On Pre Lollipop Devices

aguragorn, 2016-09-29:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.1`
 - Device OS version: `4.3`
 - Device Manufacturer: `Samsung`
 - Device Name: `Galaxy S3`

Still shot crashes on jellybean devices.
These might be related:

- http://stackoverflow.com/questions/38467680/android-getting-resourcesnotfoundexception-for-abc-ic-ab-back-material
- http://stackoverflow.com/questions/35622438/update-android-support-library-to-23-2-0-cause-error-xmlpullparserexception-bin

```
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.cormanttech.holmes.workforce/com.afollestad.materialcamera.CaptureActivity}: android.content.res.Resources$NotFoundException: File res/drawable/mcam_camera_rear.xml from drawable resource ID #0x7f0200d5
    at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2295)
    at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2349)
    at android.app.ActivityThread.access$700(ActivityThread.java:159)
...
 Caused by: android.content.res.Resources$NotFoundException: File res/drawable/mcam_camera_rear.xml from drawable resource ID #0x7f0200d5
    at android.content.res.Resources.loadDrawable(Resources.java:2974)
    at android.content.res.Resources.getDrawable(Resources.java:1558)
    at android.support.v7.widget.ResourcesWrapper.getDrawable(ResourcesWrapper.java:128)
    at android.support.v7.widget.TintResources.getDrawable(TintResources.java:45)
    at android.support.v4.content.ContextCompat.getDrawable(ContextCompat.java:346)
    at com.afollestad.materialcamera.internal.BaseCameraFragment.setImageRes(BaseCameraFragment.java:92)
    at com.afollestad.materialcamera.internal.BaseCameraFragment.onViewCreated(BaseCameraFragment.java:110)
    at com.afollestad.materialcamera.internal.CameraFragment.onViewCreated(CameraFragment.java:101)
    at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:917)
    at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1075)
    at android.app.BackStackRecord.run(BackStackRecord.java:682)
    at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:1455)
...
 Caused by: org.xmlpull.v1.XmlPullParserException: Binary XML file line #1: invalid drawable tag vector
    at android.graphics.drawable.Drawable.createFromXmlInner(Drawable.java:933)
    at android.graphics.drawable.Drawable.createFromXml(Drawable.java:873)
    at android.content.res.Resources.loadDrawable(Resources.java:2970)
    at android.content.res.Resources.getDrawable(Resources.java:1558) 
    at android.support.v7.widget.ResourcesWrapper.getDrawable(ResourcesWrapper.java:128) 
    at android.support.v7.widget.TintResources.getDrawable(TintResources.java:45) 
    at android.support.v4.content.ContextCompat.getDrawable(ContextCompat.java:346) 
    at com.afollestad.materialcamera.internal.BaseCameraFragment.setImageRes(BaseCameraFragment.java:92) 
    at com.afollestad.materialcamera.internal.BaseCameraFragment.onViewCreated(BaseCameraFragment.java:110) 
    at com.afollestad.materialcamera.internal.CameraFragment.onViewCreated(CameraFragment.java:101) 
    at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:917) 
    at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1075) 
    at android.app.BackStackRecord.run(BackStackRecord.java:682) 
    at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:1455) 
    at android.app.Activity.performStart(Activity.java:5381) 
...
```

afollestad, 2016-09-29:
You need to enable `vectorDrawables.useSupportLibrary = true` in your default Gradle config.

closed 2016-09-29 by afollestad

aguragorn, 2016-09-29:
I had it enabled already. Anyway I got it to work by adding
```java
static
{   AppCompatDelegate.setCompatVectorFromResourcesEnabled(true);
}
```
To my main application activity.
It would be nice though if devs wouldn't need to add this piece of code, especially since it is not in the documentations.

UglyTroLL, 2016-10-04:
@aguragorn I'm also having the same problem.
What did you mean by 'main application activity'? The startup activity?

aguragorn, 2016-10-04:
Any class that gets called first when your application starts. The start up activity should do since setCompatVectorFromResourcesEnabled is static anyway.
In my app however I have a class declared in
```xml
<application android:name="MainClass" ...
```
That's where I put it.

JehandadK, 2016-10-06:
When I made your fix I have another crash while capturing video

https://github.com/afollestad/material-camera/issues/119#issuecomment-252029220

jjerome00, 2016-10-11:
I was able to get it to work on API-19 devices by doing both:

1. Adding ` vectorDrawables.useSupportLibrary = true ` to my app's gradle file in the defaultConfig section
2. ` static { AppCompatDelegate.setCompatVectorFromResourcesEnabled(true); } ` in my Application class

If this solution proves to work for everyone, you might want to mention it in the project's readme.

UglyTroLL, 2016-10-11:
@jjerome00 Thanks! But unfortunately, it doesnt work for me :(