afollestad/material-camera #111: Cannot Flip between front and rear camera in still capture mode

cpritchard, 2016-09-29:
- [X] I have verified there are no duplicate active or recent bugs, questions, or requests
- [X] I have verified that I am using the latest version of Material Camera.
- [X] I have given my issue a non-generic title.

 - Material Camera version: `0.4.1`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `LG`
 - Device Name: `Nexus 5`

 - Device OS version: `6.0`
 - Device Manufacturer: `LG`
 - Device Name: `G4`

```java.lang.IllegalStateException: CameraDevice was already closed
                                                       at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:1997)
                                                       at android.hardware.camera2.impl.CameraDeviceImpl.submitCaptureRequest(CameraDeviceImpl.java:844)
                                                       at android.hardware.camera2.impl.CameraDeviceImpl.setRepeatingRequest(CameraDeviceImpl.java:899)
                                                       at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:236)
                                                       at com.afollestad.materialcamera.internal.Camera2Fragment.onPreferencesUpdated(Camera2Fragment.java:678)
                                                       at com.afollestad.materialcamera.internal.BaseCameraFragment.invalidateFlash(BaseCameraFragment.java:422)
                                                       at com.afollestad.materialcamera.internal.BaseCameraFragment.onFlashModesLoaded(BaseCameraFragment.java:165)
                                                       at com.afollestad.materialcamera.internal.Camera2Fragment.openCamera(Camera2Fragment.java:631)
                                                       at com.afollestad.materialcamera.internal.BaseCameraFragment.onClick(BaseCameraFragment.java:387)
                                                       at com.afollestad.materialcamera.internal.Camera2Fragment.onClick(Camera2Fragment.java:73)
                                                       at android.view.View.performClick(View.java:5204)
                                                       at android.view.View$PerformClick.run(View.java:21153)
                                                       at android.os.Handler.handleCallback(Handler.java:739)
                                                       at android.os.Handler.dispatchMessage(Handler.java:95)
                                                       at android.os.Looper.loop(Looper.java:148)
                                                       at android.app.ActivityThread.main(ActivityThread.java:5417)
                                                       at java.lang.reflect.Method.invoke(Native Method)
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
```
 
###### Reproduction Steps

1. Git clone project
2. Launch Sample project on phone
3. Click "Launch Camera For Stillshot" (or fragment version)
3. Click Flip camera button
4. Crash

###### Expected Result

To flip camera instead of crash

###### Actual Result

Crash

tylerkrupicka, 2016-10-01:
I get the same stack trace with a Nexus 6p on 7.0.0.

cpritchard, 2016-10-03:
With the latest commits I still get the messages but they are no longer fatal and the camera changes as expected just with a small delay.

tylerkrupicka, 2016-10-15:
I can no longer reproduce this issue on 0.4.2 of material-camera on Nexus 6p. Have not been able to reproduce on Samsung devices or HTC One M8 either.

closed 2016-10-18 by afollestad