afollestad/material-camera #112: Stillshot capture preview is horizontally mirrored and doesn't update on retry

cpritchard, 2016-09-29:
- [X] I have verified there are no duplicate active or recent bugs, questions, or requests
- [X] I have verified that I am using the latest version of Material Camera.
- [X] I have given my issue a non-generic title.

 - Material Camera version: `0.4.1`
 - Device OS version: `6.0.1`
 - Device Manufacturer: `LG`
 - Device Name: `Nexus 5`
 - Device OS version: `6.0`
 - Device Manufacturer: `LG`
 - Device Name: `G4`
 
###### Reproduction Steps

1. Git clone project
2. Launch Sample project on phone
3. Click "Launch Camera For Stillshot" (or fragment version)
3. Capture Image
4. View Captured Preview (mirrored issue)
5. Click "Retry"
6. Take new picture
7. View Captured Preview (not updated)

###### Expected Results
Captured image preview should match live preview
Captured image preview should updated on a retried still capture

###### Actual Result
Captured image preview never updates from the first capture, and is horizontally mirrored

cpritchard, 2016-10-03:
Now that it no longer crashes when flipping between front and rear camera I noticed this only happens with the front facing camera.

tylerkrupicka, 2016-10-16:
This is more of an implementation detail than a bug. Horizontally mirroring the front facing camera is actually the default behavior of the google camera app on my Nexus 6p. It comes down to what is the preferred output in that case. I personally like it matching the preview when the button was pressed.

closed 2016-10-18 by afollestad