afollestad/material-camera #113: In stillshot mode, the front/rear switch button goes away after clicking retry

afollestad, 2016-09-29:
Self explanatory.

mowcixo, 2016-09-30:
That happens in video mode too. I'm trying to fix it though.

mowcixo, 2016-09-30:
I have no clue... in fact if you tap in the area where the button should be, it works and suddenly it appears.

afollestad, 2016-09-30:
I'm guessing it's similar to the logic that I fixed which was preventing the flash toggle from showing if the camera started while showing the rear camera (instead of the front): https://github.com/afollestad/material-camera/commit/8a83ce37aee8a2b1e3ae8f9be3c8d777087be26b

tylerkrupicka, 2016-10-15:
@afollestad @mowcixo: @sugarmanz and I have fixed this in https://github.com/sugarmanz/material-camera/commit/9e865c33569bac6517090e881794277dfcba7d92. We will document and test before submitting this patch. The problem was in the openCamera() function, when determining the image drawable.

closed 2016-10-18 by afollestad