afollestad/material-camera #119: Crash: "invalid drawable tag vector" on file "res/drawable/evp_action_restart.xml"

UglyTroLL, 2016-10-02:
- [x ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x ] I have verified that I am using the latest version of Material Camera.
- [x ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.2`
 - Device OS version: `4.4`
 - Device Manufacturer: `Motorola`
 - Device Name: `MotoG`

Also, please wrap Java with correct syntax highlighting.

```
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.xxx.staging/com.afollestad.materialcamera.CaptureActivity}:android.content.res.Resources$NotFoundException: File res/drawable/mcam_camera_rear.xml from drawable resource ID #0x7f0200c4
                                                                                       at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2209)
                                                                                       at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2269)
                                                                                       at android.app.ActivityThread.access$800(ActivityThread.java:139)
                                                                                       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1210)
                                                                                       at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                       at android.os.Looper.loop(Looper.java:136)
                                                                                       at android.app.ActivityThread.main(ActivityThread.java:5102)
                                                                                       at java.lang.reflect.Method.invokeNative(Native Method)
                                                                                       at java.lang.reflect.Method.invoke(Method.java:515)
                                                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:779)
                                                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:595)
                                                                                       at dalvik.system.NativeStart.main(Native Method)
                                                                                    Caused by: android.content.res.Resources$NotFoundException: File res/drawable/mcam_camera_rear.xml from drawable resource ID #0x7f0200c4
                                                                                       at android.content.res.Resources.loadDrawable(Resources.java:2099)
                                                                                       at android.content.res.Resources.getDrawable(Resources.java:700)
                                                                                       at android.support.v7.widget.ResourcesWrapper.getDrawable(ResourcesWrapper.java:128)
                                                                                       at android.support.v7.widget.TintResources.getDrawable(TintResources.java:45)
                                                                                       at android.support.v4.content.ContextCompat.getDrawable(ContextCompat.java:346)
                                                                                       at com.afollestad.materialcamera.internal.BaseCameraFragment.setImageRes(BaseCameraFragment.java:99)
                                                                                       at com.afollestad.materialcamera.internal.BaseCameraFragment.onViewCreated(BaseCameraFragment.java:117)
                                                                                       at com.afollestad.materialcamera.internal.CameraFragment.onViewCreated(CameraFragment.java:101)
                                                                                       at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:904)
                                                                                       at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1062)
                                                                                       at android.app.BackStackRecord.run(BackStackRecord.java:684)
                                                                                       at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:1447)
                                                                                       at android.app.Activity.performStart(Activity.java:5257)
                                                                                       at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2182)
                                                                                       at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2269) 
                                                                                       at android.app.ActivityThread.access$800(ActivityThread.java:139) 
                                                                                       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1210) 
                                                                                       at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                                                       at android.os.Looper.loop(Looper.java:136) 
                                                                                       at android.app.ActivityThread.main(ActivityThread.java:5102) 
                                                                                       at java.lang.reflect.Method.invokeNative(Native Method) 
                                                                                       at java.lang.reflect.Method.invoke(Method.java:515) 
                                                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:779) 
                                                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:595) 
                                                                                       at dalvik.system.NativeStart.main(Native Method) 
                                                                                    Caused by: org.xmlpull.v1.XmlPullParserException: Binary XML file line #1: invalid drawable tag vector
                                                                                       at android.graphics.drawable.Drawable.createFromXmlInner(Drawable.java:933)
                                                                                       at android.graphics.drawable.Drawable.createFromXml(Drawable.java:877)
                                                                                       at android.content.res.Resources.loadDrawable(Resources.java:2095)
                                                                                       at android.content.res.Resources.getDrawable(Resources.java:700) 
                                                                                       at android.support.v7.widget.ResourcesWrapper.getDrawable(ResourcesWrapper.java:128) 
                                                                                       at android.support.v7.widget.TintResources.getDrawable(TintResources.java:45) 
                                                                                       at android.support.v4.content.ContextCompat.getDrawable(ContextCompat.java:346) 
                                                                                       at com.afollestad.materialcamera.internal.BaseCameraFragment.setImageRes(BaseCameraFragment.java:99) 
                                                                                       at com.afollestad.materialcamera.internal.BaseCameraFragment.onViewCreated(BaseCameraFragment.java:117) 
[... cut ...]
[... 10 lines cut ...]
```

I'm using the same version of support library as material-camera:
```
compile 'com.android.support:support-v4:24.2.1'
compile 'com.android.support:support-v13:24.2.1'
compile 'com.android.support:appcompat-v7:24.2.1'
```

Also have this in my build.gradle:
```
vectorDrawables.useSupportLibrary = true
```
 
###### Reproduction Steps

1. Upgrade library version from 0.3.2 to 0.4.2
2. Try starting CaptureActivity
3. Boom

###### Expected Result
Not crash

###### Actual Result
Crash

JehandadK, 2016-10-06:
Is there a fix for this, I tried replacing the drawable using iconRestart but no good

JehandadK, 2016-10-06:
Yours I could fix using

```
static {
        AppCompatDelegate.setCompatVectorFromResourcesEnabled(true);
    }
```

But I got this:

```
E/AndroidRuntime: FATAL EXCEPTION: main
                  android.view.InflateException: Binary XML file line #46: Error inflating class ImageButton
                      at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:704)
                      at android.view.LayoutInflater.rInflate(LayoutInflater.java:746)
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:489)
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:396)
                      at com.afollestad.easyvideoplayer.EasyVideoPlayer.onFinishInflate(EasyVideoPlayer.java:743)
                      at android.view.LayoutInflater.rInflate(LayoutInflater.java:754)
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:489)
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:396)
                      at com.afollestad.materialcamera.internal.PlaybackVideoFragment.onCreateView(PlaybackVideoFragment.java:84)
                      at android.app.Fragment.performCreateView(Fragment.java:1699)
                      at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:885)
                      at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1057)
                      at android.app.BackStackRecord.run(BackStackRecord.java:682)
                      at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:1435)
                      at android.app.FragmentManagerImpl$1.run(FragmentManager.java:441)
                      at android.os.Handler.handleCallback(Handler.java:725)
                      at android.os.Handler.dispatchMessage(Handler.java:92)
                      at android.os.Looper.loop(Looper.java:137)
                      at android.app.ActivityThread.main(ActivityThread.java:5041)
                      at java.lang.reflect.Method.invokeNative(Native Method)
                      at java.lang.reflect.Method.invoke(Method.java:511)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:793)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:560)
                      at dalvik.system.NativeStart.main(Native Method)
                   Caused by: android.content.res.Resources$NotFoundException: File res/drawable/evp_action_restart.xml from drawable resource ID #0x7f02008c
                      at android.content.res.Resources.loadDrawable(Resources.java:1958)
                      at android.content.res.TypedArray.getDrawable(TypedArray.java:601)
                      at android.widget.ImageView.<init>(ImageView.java:120)
                      at android.widget.ImageButton.<init>(ImageButton.java:87)
                      at android.support.v7.widget.AppCompatImageButton.<init>(AppCompatImageButton.java:57)
                      at android.support.v7.widget.AppCompatImageButton.<init>(AppCompatImageButton.java:53)
                      at android.support.v7.app.AppCompatViewInflater.createView(AppCompatViewInflater.java:118)
                      at android.support.v7.app.AppCompatDelegateImplV9.createView(AppCompatDelegateImplV9.java:1013)
                      at android.support.v7.app.AppCompatDelegateImplV9.onCreateView(AppCompatDelegateImplV9.java:1072)
                      at android.support.v4.view.LayoutInflaterCompatHC$FactoryWrapperHC.onCreateView(LayoutInflaterCompatHC.java:44)
                      at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:675)
                      at android.view.LayoutInflater.rInflate(LayoutInflater.java:746) 
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:489) 
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:396) 
                      at com.afollestad.easyvideoplayer.EasyVideoPlayer.onFinishInflate(EasyVideoPlayer.java:743) 
                      at android.view.LayoutInflater.rInflate(LayoutInflater.java:754) 
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:489) 
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:396) 
                      at com.afollestad.materialcamera.internal.PlaybackVideoFragment.onCreateView(PlaybackVideoFragment.java:84) 
                      at android.app.Fragment.performCreateView(Fragment.java:1699) 
                      at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:885) 
                      at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1057) 
                      at android.app.BackStackRecord.run(BackStackRecord.java:682) 
                      at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:1435) 
                      at android.app.FragmentManagerImpl$1.run(FragmentManager.java:441) 
                      at android.os.Handler.handleCallback(Handler.java:725) 
                      at android.os.Handler.dispatchMessage(Handler.java:92) 
                      at android.os.Looper.loop(Looper.java:137) 
                      at android.app.ActivityThread.main(ActivityThread.java:5041) 
                      at java.lang.reflect.Method.invokeNative(Native Method) 
                      at java.lang.reflect.Method.invoke(Method.java:511) 
[... cut ...]
                      at android.app.ActivityThread.main(ActivityThread.java:5041) 
                      at java.lang.reflect.Method.invokeNative(Native Method) 
                      at java.lang.reflect.Method.invoke(Method.java:511) 
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:793) 
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:560) 
                      at dalvik.system.NativeStart.main(Native Method) 
```

aguragorn, 2016-10-08:
setCompatVectorFromResourcesEnabled worked for me, I no longer get vector based exceptions. 
I stopped using MeterialCamera for now, there are a issues that I won't be good for a consumer facing app.

UglyTroLL, 2016-10-09:
@aguragorn thanks, are there any alternative video recorder out there?

aguragorn, 2016-10-10:
I am using the default camera app for now. I found this however and might worth looking at https://github.com/commonsguy/cwac-cam2

JehandadK, 2016-10-11:
setCompatVectorFromResourcesEnabled  works, solves a bit of the problem. But not in API 16.

UglyTroLL, 2016-10-11:
@JehandadK Thanks, what com.android.support:appcompat-v7 version you are using?

JehandadK, 2016-10-12:
24.2.+ is my version. It doesn't matter. The fix works for v18+. AFAIK

closed 2016-10-17 by afollestad

ruha9005, 2016-10-26:
@afollestad I also have this problem in my HTC One (Android 4.4). Why did you close this issue?

mowcixo, 2016-10-26:
It's fixed by 8cd8c477b783e4c5dda7411d41a987ac570dfbee but there is not tag yet for it, you should use master by now.

ruha9005, 2016-10-27:
@mowcixo Thanks. I compile MC from this commit and it solves my problem.

commit 8cd8c477 (2016-10-17): Resolved #119