afollestad/material-camera #121: Flipping device orientation can cause the preview image to be flipped upside down

mdolnik, 2016-10-08:
- [X] I have verified there are no duplicate active or recent bugs, questions, or requests
- [X] I have verified that I am using the latest version of Material Camera.
- [X] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.2`
 - Device OS version: `6.0.1 / 4.0.4 / 4.1.2`
 - Device Manufacturer: `LG / Sony / Unknown`
 - Device Name: `Nexus 5 / Xperia LT15i / ET-Q8`
 
###### Reproduction Steps

1. Run Sample app
2. Open either Video or Stillshot mode
3. Have device in landscape mode
4. Flip the device 180 so it's in landscape mode the other way (Flip quickly enough so that it doesn't reset the UI to portrait mode in between)

###### Expected Result

The preview image isn't upside down

###### Actual Result

The preview image is upside down

###### Other details

As mentioned in the Reproduction Steps, you have to flip it quick enough so it doesn't reset the UI to portrait mode in between. If the UI goes to portrait mode in between, it works as expected.

A couple things I noticed when you take a picture / video when it's "upside down":
- Stillshot mode with rear camera has the review image / output file in the correct orientation
- Stillshot mode with front camera has the review image / output file upside down
- Video mode with front/rear camera has the review video / output file upside down

tylerkrupicka, 2016-10-29:
I can reproduce this on the Nexus 6p and Galaxy S6. Thanks for the detailed bug report @mdolnik. If this doesn't get fixed soon I can take a look when I have some free time.

tylerkrupicka, 2016-11-10:
I had a moment to start debugging this: 

It looks like `setCameraDisplayOrientation()` is not being called when the orientation changes from 270 to 90. It is called when going between 0-90, 0-270 etc. I can take a closer look this weekend.

jeiselvan, 2016-11-10:
Hi,

I had the same bug as mentioned above. After switching between "normal" and "upside down" landscape mode, the camera preview will be upside down.
Reproduction steps:
1. Make sure phone rotation is unlocked.
2. Press Take new picture in a conversation.
3. Put the phone in "normal" landscape mode (home button to the right).
4. Rotate the phone 180 degrees counter-clockwise.
5. The controls will be correctly rotated, but the actual camera preview will be upside down.

It would be great if this bug gets fixed.

mdolnik, 2016-11-11:
Note that `mDisplayOrientation` is set in `setCameraDisplayOrientation()` for `CameraFragment.java` and it is set in `openCamera()` for `CameraFragment2.java`
Not that it makes too much of a difference because I believe they both suffer from the same issue.

According to http://stackoverflow.com/a/13306633 it seems that Fragments are re-created or re-initialized every time a screen rotation happens, in this case landscape to portrait or vise versa.

But I assume Android tries to be efficient by not re-initializing when rotating landscape to landscape because they both have the same dimensions.

According to http://stackoverflow.com/questions/9909037/how-to-detect-screen-rotation-through-180-degrees-from-landscape-to-landscape it seems to confirm that it's an issue of the fragment not getting updated for this scenario.

Their proposals are to add an `OrientationEventListener` to periodically check the current rotation.
Seems like an ugly solution, but I can't seem to find any other existing solutions.

**So with that in mind I found this will fix it, but I'm not sure if this is the best approach.**
Add this to the top of **`BaseCameraFragment.java`**
```
protected OrientationEventListener mOrientationEventListener;
private int mLastRotation;
```
Add this to the top of **`BaseCameraFragment.java onCreateView()`**
```
if (mOrientationEventListener == null) {
    mOrientationEventListener = new OrientationEventListener(getActivity(), SensorManager.SENSOR_DELAY_UI) {
        @Override
        public void onOrientationChanged(int angle) {
            // Check if current rotation is landscape but different than before
            int rotation = getActivity().getWindowManager().getDefaultDisplay().getRotation();
            if ((rotation == Surface.ROTATION_90 || rotation == Surface.ROTATION_270) && rotation != mLastRotation) {
                Log.d("OrientationEvent", "angle changed" + rotation);
                // Update rotation
                mLastRotation = rotation;
                // Reset camera
                cleanup();
                openCamera();
            }
        }
    };
}
```

I'm new to android development so forgive me if I am speaking gibberish or if I'm misleading.

mdolnik, 2016-11-11:
One issue I've noticed with the code change above is if you take a video and end the video "Upside down" it crashes. I'm sure there's other potential issues, but I'm just throwing suggestions out there.

tylerkrupicka, 2016-11-11:
@mdolnik thanks for the research

rinku9900, 2017-02-09:
Is this issue fixed? I am experiencing the same issue. After photo is taken, an image is flipped. I am using version com.afollestad:material-camera:0.4.4.

mdolnik, 2017-03-03:
Has any work been done on this? Should I push my commit of what I have so far?

ejezisek, 2017-04-08:
I did it slightly differently as per: http://stackoverflow.com/questions/19577299/android-camera-preview-stretched

This was needed because while your fix above fixed it on several phones it seemed like the issue was still present on a Samsung Galaxy S5.

I made some changes to support the rotation if a video is being taken.  Please let me know if anyone has any issues with the pull request I made.  Currently it is failing due to using Android Build tools 25 in the compilation.

AjmalZia, 2017-05-07:
Hi, 
Is this issue fixed yet ?
I still face this issue on my nexus 5 , with the library code.