afollestad/material-camera #122: Screen Overlay problem in Marshmallow

shafqatjamil, 2016-10-10:
I am using your library for video recording its working fine but in my client whe she tries to record video its not working app says you need to turn off screen overlay. following is the link of video for problem
https://youtu.be/q2m_LNDdq2g

please help me I am stuck in this problem and i could not solve it i tried and searched a lot

aguragorn, 2016-10-11:
She would need to close whatever app she has that overlays on the screen, for example fb messenger's floating chat bubble.

I would think this is a general marahmallow issue, especially when granting pernissions. An overlaying app may have control on anything beneath

shafqatjamil, 2016-10-13:
yes that was it thanx for helping 👍

closed 2016-10-17 by afollestad