afollestad/material-camera #123: Cannot get v4.1.3 through gradle build

gokceyucel, 2016-10-12:
- [ ] I have verified there are no duplicate active or recent bugs, questions, or requests
- [ ] I have verified that I am using the latest version of Material Camera.
- [ ] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.4.1`
 - Device OS version: `6.0`
 - Device Manufacturer: `Samsung`
 - Device Name: `Galaxy S3`

Also, please wrap Java with correct syntax highlighting.

```java
Start: Mon Sep 26 11:15:24 UTC 2016
Git:
0.4.1
commit 03a0b9fdd162cd3f935f0aabd727e1701b2f2b1c
Author: Aidan Follestad 
Date:   Sun Sep 25 16:47:27 2016 -0500

    0.4.1

submodule status:

Run gradle build
Gradle build script
Found gradle version: 3.1.
Using sdkman gradle
Picked up JAVA_TOOL_OPTIONS: -Dfile.encoding=UTF-8

------------------------------------------------------------
Gradle 3.1
------------------------------------------------------------

Build time:   2016-09-19 10:53:53 UTC
Revision:     13f38ba699afd86d7cdc4ed8fd7dd3960c0b1f97

Groovy:       2.4.7
Ant:          Apache Ant(TM) version 1.9.6 compiled on June 29 2015
JVM:          1.8.0_101 (Oracle Corporation 25.101-b13)
OS:           Linux 3.14.32-xxxx-grs-ipv6-64 amd64

0m0.700s
Getting a list of gradle tasks
Picked up JAVA_TOOL_OPTIONS: -Dfile.encoding=UTF-8

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':library'.
> Could not resolve all dependencies for configuration ':library:_debugPublishCopy'.
   > Could not find com.afollestad:easyvideoplayer:0.2.13.
     Required by:
         project :library

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

WARNING:
Gradle 'install' task not found. Please add the 'maven' or 'android-maven' plugin.
See the documentation and examples: https://jitpack.io/docs/

Found android library build file in library
Running: gradle clean -Pgroup=com.github.afollestad -Pversion=0.4.1 install
Picked up JAVA_TOOL_OPTIONS: -Dfile.encoding=UTF-8
Download https://jcenter.bintray.com/com/github/dcendents/android-maven-gradle-plugin/1.5/android-maven-gradle-plugin-1.5.pom
Download https://jcenter.bintray.com/com/github/dcendents/android-maven-gradle-plugin/1.5/android-maven-gradle-plugin-1.5.jar
Gradle version Gradle 3.1
Version: 0.4.1
Build dir: /home/jitpack/build/library/build
File /home/jitpack/.android/repositories.cfg could not be loaded.

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':library'.
> Could not resolve all dependencies for configuration ':library:_debugPublishCopy'.
   > Could not find com.afollestad:easyvideoplayer:0.2.13.
     Required by:
         project :library

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 4.119 secs
Gradle exit code 1
EXIT_CODE=1
2016-09-26T11:15:51.382033122Z
Exit code: 1
No build artifacts found
```
 
###### Reproduction Steps

1. add ```compile 'com.afollestad:material-camera:0.2.13'``` to your build.gradle
2. build

###### Expected Result
Successfull build

###### Actual Result
Failed build

Here is proof: https://jitpack.io/com/github/afollestad/material-camera/0.4.1/build.log

gokceyucel, 2016-10-12:
I think the problem is in this line: 
```Could not find com.afollestad:easyvideoplayer:0.2.13.```
Maybe it should be 0.2.3 according to this -> https://github.com/afollestad/easy-video-player/releases

afollestad, 2016-10-17:
You're using the wrong repository. Use jCenter.

closed 2016-10-17 by afollestad