afollestad/material-camera #128: ImageUtil calculateInSampleSize using a lot of memory

jjerome00, 2016-10-13:
- [x] I have verified there are no duplicate active or recent bugs, questions, or requests
- [x] I have verified that I am using the latest version of Material Camera.
- [x] I have given my issue a non-generic title.

###### Include the following:
 - Material Camera version: `0.2.13`
 - Device OS version: `4.4.2` KitKat API 19
 - Device Manufacturer: `Samsung`
 - Device Name: `SAMSUNG-SM-G900A : Galaxy S5`

I was noticing a huge memory spike whenever the library was displaying the picture in StillshotPreviewFragment.  I followed the logic back to the ImageUtil class, specifically the calculateInSampleSize() method.

I noticed this method is very similar to the one that Google gives as an example for proper bitmap scaling.  I had bitmap memory issues in another spot in my app, and I found a different way of calculating the inSampleSize that seems to make better use of memory.

Instead of making any assumptions and just creating a pull request out of the blue, I decided to ask here to see if my initial research (watching the Android Monitor memory graph) had any merit.

Existing (ImageUtil.java):
```java
    private static int calculateInSampleSize(
            BitmapFactory.Options options, int reqWidth, int reqHeight, int inSampleSize) {
        // Raw height and width of image
        final int height = options.outHeight;
        final int width = options.outWidth;

        if (height > reqHeight || width > reqWidth) {

            final int halfHeight = height / 2;
            final int halfWidth = width / 2;

            // Calculate the largest inSampleSize value that is a power of 2 and keeps both
            // height and width larger than the requested height and width.
            while ((halfHeight / inSampleSize) > reqHeight
                    && (halfWidth / inSampleSize) > reqWidth) {
                inSampleSize *= 2;
            }
        }

        return inSampleSize;
    }
```

New:
```java
private static int calculateInSampleSize2(BitmapFactory.Options options, int reqWidth, int reqHeight) {

        // Raw height and width of image
        final int height = options.outHeight;
        final int width = options.outWidth;
        int inSampleSize = 1;

        if (height > reqHeight || width > reqWidth) {
            // Calculate ratios of height and width to requested height and width
            final int heightRatio = Math.round((float) height / (float) reqHeight);
            final int widthRatio = Math.round((float) width / (float) reqWidth);

            // Choose the smallest ratio as inSampleSize value, this will guarantee
            // a final image with both dimensions larger than or equal to the
            // requested height and width.
            inSampleSize = heightRatio < widthRatio ? heightRatio : widthRatio;
        }
        return inSampleSize;
    }
```
I noticed the caller (ImageUtil.getRotatedBitmap) does this little recursion trick where it keeps calling itself trying a higher scale number if there is an out of memory error.  The new method ignores that value for now.

After making this change I noticed quite a dip in memory used when displaying the StillshotPreviewFragment.

I cannot take any credit for this piece of code.  I found it on StackOverflow here:
[http://stackoverflow.com/questions/17839388/creating-a-scaled-bitmap-with-createscaledbitmap-in-android/17839597#17839597](http://stackoverflow.com/questions/17839388/creating-a-scaled-bitmap-with-createscaledbitmap-in-android/17839597#17839597)

tylerkrupicka, 2016-10-15:
I'm seeing some similar results as you, thanks for the code sample. This is implemented in https://github.com/sugarmanz/material-camera/commit/f6db69aaa0503353c17e6a58f6e9a799c46ccecc with some fixes for rotated images. We were seeing a pretty significant drop in memory using this method over the Google one, but we won't submit a PR until we do some testing across all the devices we can find. Good work!

afollestad, 2016-10-17:
It would've been easier if you submitted a PR ;)

commit da463f17 (2016-10-17): Updated the calculateInSampleSize method, resolves #128
closed 2016-10-17 by afollestad